adafruit/circuitpython #8098: Byte format when using bytearray() with .append()

garciaalex335, 2023-06-20:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.5 on 2023-03-31; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
message = bytearray()
        print("CHECKING PACKET")
        
        message.append(FRAMESTART) # 0xC5

        len_high = (PACKET_LEN >> 8) & 0xFF
        len_low = PACKET_LEN & 0xFF
        
        len_packed = struct.pack("B",int(105))
        #message.extend(len_packed)       
        print(len_packed)
        message.append(105)
        message.append(len_high)
        print(message)
```

### Behavior

CHECKING PACKET
b'i'
bytearray(b'\xc5i\x00')

### Description

I'm attempting to create a data to pass via UART. In trying to add a particular byte it seems to be getting formatted improperly. I would expect the prints to look like:

CHECKING PACKET
b'\x69'
bytearray(b'\xc5\x69\x00')

Uncommenting the line message.extend(len_packed) and commenting message.append(105) gives the same result.

For now I am working around it by hard coding params into the initialization message = bytearray(....)

anecdata, 2023-06-20:
Behaves the same in CPython```py
>>> x = b'\x69'
>>> x
b'i'
>>> 
>>> y = bytearray(b'\xc5\x69\x00')
>>> y
bytearray(b'\xc5i\x00')
```

dhalbert, 2023-06-20:
Python, even in bytestrings, prints the regular printing character if possible. Otherwise it prints an `\x...` escape. Note that b`\x69` and b`i` are exactly the same thing, just printed out differently. They are both one byte long, and contain a single byte whose integer value is `0x69`.

closed 2023-06-20 by dhalbert