adafruit/circuitpython #8107: Mounted filesystems don't appear over web workflow

makermelissa, 2023-06-22:
### CircuitPython version

```python
8.2.0-beta.1-12-g0da62416a-dirty
```

### Code/REPL

```python
N/A
```

### Behavior

Per @ladyada:

Mounted filesystems don't appear - i have a microSD on /sd but does not show up over wifi.

Checked via /fs/ folder on the device and it wasn't there either.

tannewt, 2023-11-30:
I've been thinking a lot about this and I think we'll need to require that the FS be backed by a native object for reading and writing blocks. MP actually allows for Python implemented block io but that would lead to running Python functions at anytime while other user code runs.

ladyada, 2023-11-30:
can we do that easier if we requre using sdcardio (which is already semi-native)

tannewt, 2023-11-30:
Yup, exactly. Using the native sdcardio or sdio rather than the Python sd card library would be required.

PR #8699 (2024-01-25): Make SD cards available over web workflow
This changes storage.mount() to require that a mount point exist on the parent file system.

A bug in background tasks is also fixed where the function parameter is cleared on pending callbacks during "reset".

Disk usage is shown on the directory listing and changes based on the mounted file system. Writable is also loaded per-directory.

Fixes #8108. Fixes #8690. Fixes #8107.
commit 25e862d1 (2024-01-25): Make SD cards available over web workflow

This changes storage.mount() to require that a mount point exist
on the parent file system.

A bug in background tasks is also fixed where the function
parameter is cleared on pending callbacks during "reset".

Disk usage is shown on the directory listing and changes based on
the mounted file system. Writable is also loaded per-directory.

Fixes #8108. Fixes #8690. Fixes #8107.
closed 2024-01-25 by tannewt