adafruit/circuitpython #8119: docs: Missing references to ReadableBuffer, WritableBuffer etc. circuitpython_typing types

nomadbyte, 2023-06-25:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22
```

### Code/REPL

```python
# Documentation issue; Not in Code/REPL
```

### Behavior

```python
# Documentation issue; Not in Code/REPL
```

### Description

This is a CircuitPython API documentation issue.

In the latest and releases after 7.2.x, the API reference docs no longer have links for the `circuitpython_typing` types.

For example, in docs for the latest version see `circuitpython_typing.WriteableBuffer` link missing in [`usb_midi.PortIn.readinto`](https://docs.circuitpython.org/en/latest/shared-bindings/usb_midi/index.html#usb_midi.PortIn.readinto).

Yet, in docs for the 7.2.x, that same type had a valid link, see [`usb_midi.PortIn.readinto` 7.2.x](https://docs.circuitpython.org/en/7.2.x/shared-bindings/usb_midi/index.html#usb_midi.PortIn.readinto). The `circuitpython_typing.WriteableBuffer` 7.2.x  link is pointing at [https://docs.circuitpython.org/en/7.2.x/shared-bindings/circuitpython_typing/index.html#circuitpython_typing.WriteableBuffer](https://docs.circuitpython.org/en/7.2.x/shared-bindings/circuitpython_typing/index.html#circuitpython_typing.WriteableBuffer).

In the docs for the latest and 8.0.x, 7.3.x versions the respective links are 404, for example:

[https://docs.circuitpython.org/en/8.0.x/shared-bindings/circuitpython_typing/index.html#circuitpython_typing.WriteableBuffer](https://docs.circuitpython.org/en/8.0.x/shared-bindings/circuitpython_typing/index.html#circuitpython_typing.WriteableBuffer).

This issue applies to all `circuitpython_typing` types.

tekktrik, 2023-07-14:
It may make more sense to document it in the `cirucitpython_typing`, and then link to that from the main documentation.

kbsriram, 2023-07-30:
The "easy" solution seemed to be to add a reference to the `circuitpython_typing`  repo around here:
https://github.com/adafruit/circuitpython/blob/8050b07a687277cb277e4c13de83cfdc076682d5/conf.py#L444

However, there's an additional long-standing Sphinx issue, which appears to have trouble with `TypeAlias` definitions (see https://github.com/sphinx-doc/sphinx/issues/8934)

The symptom: the Sphinx doc generator looks for a `class` definition for a `TypeAlias`ed definition like 
 `circuitpython_typing.ReadableBuffer` but the online repository tags it as `data` as can be seen when currently running
```
python -m sphinx.ext.intersphinx https://docs.circuitpython.org/projects/adafruit-circuitpython-typing/en/latest/objects.inv
```
Could one workaround be to intercept missing references, and retry it with a `data` match? I'll add a PR with the proposed approach.

closed 2023-07-30 by jepler