adafruit/circuitpython #8120: 8.2.0-rc.0, no echo on REPL on ODroid Go

fovea1959, 2023-06-25:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-rc.0 on 2023-06-23; Hardkernel ODroid Go with ESP32
```

### Code/REPL

```python
None
```

### Behavior

After a powerup and connection via the UART, characters typed in the REPL show up on the screen, but are not echoed to the host PC.

Issuing commands `import os` `os.listdir()` causes screen garbage.

`import wifi` still hangs.

### Additional information

Is there a way to disable the onboard screen from echoing the REPL. This and #8069 smell like some sort of bad memory access when scrolling the screen, and disabling the screen would help me debug that.

tannewt, 2023-06-26:
You can disable the display by calling `displayio.release_displays()`. With a build you can also comment out all of the display init in the `board_init` function.

dhalbert, 2024-03-08:
Worth retesting with 9.0.0,but this might be a bunch of different issues, such as UART setup troubles and/or display troubles.