adafruit/circuitpython #8124: ESP32S3 crash and print nothing with IRRemote DEMO.

StreakingJerry, 2023-06-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-23; ESP32-S3-Box-Lite with ESP32S3
```

### Code/REPL

```python
# Circuit Playground Express Demo Code
# Adjust the pulseio 'board.PIN' if using something else
import pulseio
import board
import adafruit_irremote

pulsein = pulseio.PulseIn(board.G38, maxlen=120, idle_state=True)
decoder = adafruit_irremote.GenericDecode()

while True:
    pulses = decoder.read_pulses(pulsein)
    print("Heard", len(pulses), "Pulses:", pulses)
    try:
        code = decoder.decode_bits(pulses)
        print("Decoded:", code)
    except adafruit_irremote.IRNECRepeatException:  # unusual short code!
        print("NEC repeat!")
    except adafruit_irremote.IRDecodeException as e:     # failed to decode
        print("Failed to decode: ", e.args)

    print("----------------------------")
```

### Behavior

Blocking in `decoder.read_pulses(pulsein)` lead to crash.

Also, This simple demo prints nothing on the ESP32S3. I tried some debug and found that `pulseio.PulseIn` receviced nothing. However, this code works good on RP2040 with `board.GP2'.

DavePutz, 2023-06-28:
I don't have a ESP32-S3-Box-Lite to test with; but running CircuitPython 8.1.0 on an ESP32-S3-DevKitC-1-N8 I was able to see pulses correctly using the provided script. All I can suggest would be to make sure you are on the correct pin. Perhaps just using digitalio on the pin to verify? Something like:
```
import digitalio
import board
import time

pin = digitalio.DigitalInOut(board.G38)
while True:
    print(pin.value)
    time.sleep(.1)
```
Should show state changes when you use the remote.

dhalbert, 2023-10-25:
Closing due to inactivity. Please reopen if this is still an issue for you.

closed 2023-10-25 by dhalbert