adafruit/circuitpython #8131: Enhance supervisor.set_usb_identification to also be able to set USB_INTERFACE_NAME

darkyellow, 2023-06-29:
set_usb_identification allows people to currently change the manufacturer, product, vendor id and product id but it doesn't allow the changing of the usb interface name. This is currently hard coded in supervisor.mk.

The enhancement request is to also be able to change the usb interface name in the set_usb_identification method call

dhalbert, 2023-06-29:
Related to #4191. Since we are talking about the interface name for `usb_hid`, might be added to that module instead.

skz000, 2025-03-10:
It looks like HID and MIDI interface renaming were implemented without needing to rebuild CircuitPython. It would be great if USB_INTERFACE_NAME could be updated in a similar way in the future.