adafruit/circuitpython #8132: While running code in a loop, web workflow is unresponsive to Ctrl+C

makermelissa, 2023-06-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-rc.1 on 2023-06-27; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
import time

while True:
    print("Hello World!")
    time.sleep(1)
```

### Behavior

Using a Web Workflow serial connection, pressing Ctrl-C does not work as expected. It occasionally works, but not most of the time.

### Description

I verified that this works as expected in 8.0.0-beta.6, so it broke along the way. I'm pretty sure it doesn't work in 8.1.0 either. I'm not sure when it stopped working.

### Additional information

See https://github.com/circuitpython/web-editor/issues/115 for additional details.

dhalbert, 2023-07-05:
Maybe should fix this in 8.2.x.

tannewt, 2023-11-28:
I think this was broken by https://github.com/adafruit/circuitpython/pull/7836. It adds a catch all break when no new connection is available that skips the websocket_background call.

PR #8668 (2023-11-28): Fix websocket input processing
It was skipped in many cases which breaks ctrl-cing running code.

Fixes #8132
commit 96b0c211 (2023-11-28): Fix websocket input processing

It was skipped in many cases which breaks ctrl-cing running code.

Fixes #8132
closed 2023-11-28 by dhalbert