adafruit/circuitpython #8134: Firmware CircuitPython 8.2.0 Release Candidate 1 broken for Feather NRF52840 Express

mindforger, 2023-06-30:
### CircuitPython version

```python
CircuitPython 8.2.0 Release Candidate 1
```

### Code/REPL

```python
no code
```

### Behavior

Circuitpython FW DFU flashing fails and is not working

### Description

download 8.1.0 and 8.2.0-rc1
plug in usb and double tal reset for DFU
move 8.2.0-rc1, see copy process work until 80% and abort connection with a file copy error (donwloaded and checked the file integrity)
move 8.1.0, see copy finish normally and circuitpython launching

### Additional information

none

dhalbert, 2023-06-30:
This is actually due to a bug in the UF2 bootloader on the board. Double-click to get to the FTHR840BOOT drive, and check the bootloader version in INFO_UF2.TXT. It is probably 0.2.6 or 0.2.9. Any version below 0.6.1 cannot load UF2's that contain more than 512kB of payload. CircuitPython 8.2.0 is more than 512kB.

To update the bootloader, follow the instructions here:
https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/update-bootloader

The Arduino method is probably the easiest if you are familiar with Arduino. It will update to 0.6.2. Once you update to 0.4.0 or higher, you can use the UF2's in https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases to self-update. Bootloader versions like 0.2.6 or 0.2.9 cannot self-update.

Regardless of which update method you choose, make sure no other program is attached to the serial port of the board while updating, or the update will not work. So only have Arduino running when burning the bootloader, and then quit Arduino completely and try reloading CircuitPython.

I reproduced your problem with some older Feather nRF52840 boards. When I updated the bootloader, they could then load 8.2.0-rc.1.

dhalbert, 2023-06-30:
I will add a prominent caveat to the release notes about this.

mindforger, 2023-07-01:
Also, in the github release page for the bootloader, there are a LOT of hex files and zip but no advice which file is for which Board, this could use some love too :)

closed 2023-07-05 by dhalbert