adafruit/circuitpython #8135: circuitpython on zephyr

mfgamma, 2023-07-01:
hi

well it would be a major boost for hi end embedded applications to have it on zephyr rtos

tannewt, 2023-07-05:
What would you want from CircuitPython that [MicroPython on Zephyr](https://github.com/micropython/micropython/tree/master/ports/zephyr) doesn't provide?

Jones14021, 2024-07-23:
My main need would be the Adafruit web libraries like the [adafruit_httpserver](https://docs.circuitpython.org/projects/httpserver/en/latest/index.html)
and the USB/web workflows.
Relying on frozen modules in Micropython for run-on-boot capability doesn't even come close to the convenience of the CircuitPython USB worflow imho.

Zephyr OS offers very easy portability and very wide hardware support. Especially **Ethernet support (for internal MACs)** is one major feature that is missing from CircuitPython at the moment (unless one wants to use the WIZNET5K chips externally).
Since adding Ethernet support to ports like the `ports/atmel-samd` (e.g. for using SAM E54 built-in GMAC peripheral) looks to be quite challenging and limited to one port at a time, using the Zepyhr OS Ethernet implementation would promise:

- easy maintenance (relying on Zephyr's HAL)
- broad hardware support
- common implementation and API for all supported devices on CircuitPython side

One hope of mine would also be that e.g. protocol support for Zigbee, Thread etc. would be far easier to achieve when relying on Zephyr OS's hardware abstraction.

tannewt, 2024-07-23:
I'd be happy to review and merge a Zephyr port of some sort. I'm still concerned that Zephyr isn't dynamic enough to build CP on top of. (Pin muxing is the main example.) I'd be happy if someone could prove me wrong.

tannewt, 2024-12-20:
I've opened #9902 so that I can edit my own top post. Thanks for opening this earlier.

closed 2024-12-20 by tannewt