adafruit/circuitpython #8138: Pico W simple AP code is boot looping

anecdata, 2023-07-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-rc.0 on 2023-06-23; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import time
import wifi

time.sleep(3)  # wait for serial
ap_ssid = "PicoWAP"
ap_password = "picow1234"
print("Stopping station mode.")
wifi.radio.stop_station()
print("Starting AP mode.")
wifi.radio.start_ap(ssid=ap_ssid, password=ap_password)
print("Access point created with SSID: {}, password: {}".format(ap_ssid, ap_password))
print("My IP address is", wifi.radio.ipv4_address_ap)
```

### Behavior

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.2.0-rc.0 on 2023-06-23; Raspberry Pi Pico W with rp2040
>>> 
>>> 
>>> import microcontroller 
>>> microcontroller.reset()

[16:20:27.279] Disconnected
[16:20:28.285] Warning: Could not open tty device (No such file or directory)
[16:20:28.285] Waiting for tty device..
[16:20:32.364] Connected

Auto-reload is off.
code.py output:
Stopping station mode.
Starting AP mode.
Access point created with SSID: PicoWAP, password: picow1234
My IP address is 192.168.4.1

Code done running.

[16:20:37.179] Disconnected
[16:20:38.181] Warning: Could not open tty device (No such file or directory)
[16:20:38.181] Waiting for tty device..
[16:20:44.267] Connected

Auto-reload is off.
code.py output:
Stopping station mode.
Starting AP mode.
Access point created with SSID: PicoWAP, password: picow1234
My IP address is 192.168.4.1

Code done running.
```
Rinse, repeat.

### Description

If a `while True:` loop is added to the end of the code, and control-C is attempted after some time, it will not crash. But sometimes cvrashes if control-C is too soon.

eightycc, 2023-11-12:
At first I was unable to reproduce this problem, but that was because pull #7101 had turned the crash into a "Stopping AP is not supported." in `common_hal_wifi_radio_stop_ap()`. Looking into the reason for that I found https://github.com/georgerobotics/cyw43-driver/issues/47 which notes that "Something needs to call cyw43_wifi_set_up with up set to false...". Doing that solves this issue and makes `wifi.stop_ap()` functional for RP2.

anecdata, 2023-11-13:
Closed by #8590

PR #8590 (2023-11-13): Implement wifi.stop_ap for RP2
Implement `wifi.stop_ap()` in RP2.

Resolves #8138.
closed 2023-11-13 by anecdata