adafruit/circuitpython #8140: Support for RTL8720D?

steveiliop56, 2023-07-04:
Hello!

Is there any work in progress or a working library for the wio terminal wifi chip (rtl8720d) or it will never be supported?

skerr92, 2023-07-04:
Given that the MicroPython isn’t currently supporting any of the rtl chips (they supported the ameba boards at some point but I don’t see it in the ports anymore), it’s probably not likely unless Realtek adds support themselves. Having worked on Realtek chips, there’s quite a bit behind NDA that could be a hurdle for being support.

tannewt, 2023-07-05:
> Is there any work in progress or a working library for the wio terminal wifi chip (rtl8720d) or it will never be supported?

No one funded by Adafruit is working on it or plans to. I'd suggest asking Seeed about it.

steveiliop56, 2023-11-25:
Yeah I guess no one will ever add support for it...

closed 2023-11-25 by steveiliop56