adafruit/circuitpython #8141: Please reopen this issue, as I have it on ESP32-S3 DevKitC N8R2, with CP 8.2.0.

dhalbert, 2023-07-06:
Please reopen this issue, as I have it on ESP32-S3 DevKitC N8R2, with CP 8.2.0.

I did apply this PR #5858 on CP 8.2.x and it solved my issue!!

_Originally posted by @casainho in https://github.com/adafruit/circuitpython/issues/5662#issuecomment-1623529340_

dhalbert, 2023-07-06:
@casainho I have created a new issue. And the new problem is that a simple `import wifi`  crashes?

So you're saying the old change from #5858 fixes your new problem? Was the change from #5858 lost? Thanks.

jepler, 2023-07-06:
the current content seems to be:
```
    config ESP_PHY_ENABLE_USB
        bool "Enable USB when phy init"
        depends on USB_OTG_SUPPORTED || IDF_TARGET_ESP32C3
```
which is different than either side of #5858.  It may have been changed by
```
commit d3d5e36548fdaf7b0d3b7bce7861986ed8080829
Author: Dan Halbert <halbert@adafruit.com>
Date:   Tue Aug 2 22:37:06 2022 -0400

    update adafruit/esp-idf for ESP32_ECO3_CACHE_LOCK_FIX fix
```
and/or
```
commit 96dd11a1986ad99029a1668cc6b88c322e2947be
Author: Scott Shawcroft <scott@adafruit.com>
Date:   Fri Mar 11 14:40:49 2022 -0800

    Update ESP-IDF to fix PinAlarm
    
    Fixes #5973
```
it's a bit hard to tell from git history when submodules are involved.

jepler, 2023-07-06:
96dd11a1986ad99029a1668cc6b88c322e2947be seems to have caused ESP_PHY_ENABLE_USB to start being defined when building for BOARD=espressif_esp32c3_devkitm_1_n4

dhalbert, 2023-07-06:
The current submodule choice for ESP-IDF is
https://github.com/adafruit/circuitpython/blob/fad86bfb27607ca362c124bcc8ecef671d3fd40f/.gitmodules#L143-L146, which is different from the older choices above (our fork, new branch)

dhalbert, 2023-07-06:
According to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp-phy-enable-usb, it is correct to have `CONFIG_ESP_PHY_ENABLE_USB=y` for ESP32-C3. When I build the board with the tip of main, that value is set in `sdkconfig`.

dhalbert, 2023-07-06:
@casainho Getting back to the original problem with ESP32-**S3** DevKitC N8R2:

I loaded the 8.2.0 .bin from https://circuitpython.org/board/espressif_esp32s3_devkitc_1_n8r2/

I am unable to cause a crash by a simple `import wifi`:
```py
Adafruit CircuitPython 8.2.0 on 2023-07-05; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
>>> 
>>> 
>>> import wifi
>>> 
```

casainho, 2023-07-06:
@dhalbert , it hangs here:

[image]

But with the modified CP as I told:

[image]

casainho, 2023-07-06:
The code with the issue is here: https://github.com/OpenSourceEBike/EBike_EScooter_modular_DIY/tree/issue_wifi/diy_main_board/firmware

And here the CP binary where the issue do not happen: https://github.com/OpenSourceEBike/EBike_EScooter_modular_DIY/blob/issue_wifi/diy_main_board/firmware/circuitpython_binary/adafruit-circuitpython-espressif_esp32s3_devkitc_1_n8r2-en_US-8.2.x-built_issue_import_wifi.bin

casainho, 2023-07-06:
Other interesting thing, is that even with original binary CP 8.2.0, the board appears as N32R8, but is is a N8R2.

[image]

jepler, 2023-07-06:
I'm sorry for creating confusion by referring to a board with *esp32-c3* when this is about a board with an *s3* microcontroller (that just happens to have a *c* in product name too)

dhalbert, 2023-07-06:
@casainho I have both the N32R8 board and the N8R2 board. If you look on the tin case of the module in the lower left, there is a stamping that includes either N32R8 or N8R2. Could you look at the the one that you have and see what it says. Also the N8R2 says ESP32-S3-WROOM-1 on the N8R2 board, and ESP32-S3-WROOM-2 on the N32R8 board.

I am not sure which line is causing the issue from the screen shots. Could you give me a source link to the line in the repo? Click on the line number to get a "...", then click on that and choose "Copy Permalink"

casainho, 2023-07-07:
> @casainho I have both the N32R8 board and the N8R2 board. If you look on the tin case of the module in the lower left, there is a stamping that includes either N32R8 or N8R2. Mine identify properly in `lsusb`. Could you look at the the one that you have and see what it says? Also the N8R2 says ESP32-S3-WROOM-1, and ESP32-S3-WROOM-2 on the N32R8 board.

lsusb says "303a:7003 Espressif ESP32-S3-DevKitC-1-N8R2", so I guess that plugin for Visual Code Studio is showing the wrong data.

Picture:
[image]

> I am not sure which line is causing the issue from the screen shots. Could you give me a source link to the line in the repo? Click on the line number to get a "...", then click on that and choose "Copy Permalink".

https://github.com/OpenSourceEBike/EBike_EScooter_modular_DIY/blob/acfe6b440dd7978ba723148e03c3cd09705e2f72/diy_main_board/firmware/escooter_fiido_q1_s/main.py#L82

But I think the issue seems to be about something that is happening in parallel, and after some time, it resets the board - and not a issue on some specific line of code.

I found this issue when I tried to change the board MAC Address with this code:
```
import wifi
wifi.radio.enabled = True
my_mac_address = [0x68, 0xb6, 0xb3, 0x01, 0xf7, 0xf2]
wifi.radio.mac_address = bytearray(my_mac_address)
```

I am using that same code on ESP32-S2, and it works as expected.

> Are you building 8.2.0, or did you try the download version? The submodules have changed a bunch of times, both in terms of URL and branch used.

I was using the 8.2.0 from the CP download page. Just now I did built my own version to add the previously mentioned changes.

dhalbert, 2023-07-07:
That photo does not look at all like https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html or the photos here: https://www.adafruit.com/product/5310. The buttons are in a completely different place. Could you give a link to the board that you purchased?

casainho, 2023-07-07:
This is the board, I bought on [Aliexpress](https://pt.aliexpress.com/i/1005004617322170.html?gatewayAdapt=glo2bra):

[image]

dhalbert, 2023-07-07:
@casainho Are you using a `settings.toml` file. If so, what is in it (fine to redact SSID and password)?

closed 2024-03-08 by dhalbert