adafruit/circuitpython #8145: Playing long wav file on sdcard with audiocore.WaveFile() will cause adafruit_feather_esp32s3_tft reboot to the boot mode

HonestQiao, 2023-07-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
import board
import audiocore
import audiobusio
import digitalio

import busio
import storage
import adafruit_sdcard

import os
import sys
import time

print("start")
spi = board.SPI()
cs = digitalio.DigitalInOut(board.RX)

print("connect")
sdcard = adafruit_sdcard.SDCard(spi, cs)

print("VfsFat")
vfs = storage.VfsFat(sdcard)

print("mount")
storage.mount(vfs, "/sd")
print("mount ok")

'''
Board 3V to breakout VIN
Board GND to breakout GND
Board A0 to breakout BCLK
Board A1 to breakout LRC
Board A2 to breakout DIN
Speaker + to screw terminal +
Speaker - to screw terminal -
'''

I2S_BCLK = board.A0
I2S_LRC = board.A1
I2S_DIN = board.A2
I2S_GAIN = board.A3

print("init i2s")
audio = audiobusio.I2SOut(I2S_BCLK, I2S_LRC, I2S_DIN)

gain = digitalio.DigitalInOut(board.A3)
gain.direction = digitalio.Direction.OUTPUT
gain.value = False

#wav_file = "/sd/music/StreetChicken.wav"
wav_file = "/sd/music/xzl2.wav"
stats = os.stat(wav_file)
filesize = stats[6]
print("filesize of %s is %d" % (wav_file, filesize))

count=0
print("play wav")
wav = audiocore.WaveFile(wav_file)
print("Playing wav file!")
audio.play(wav)
while audio.playing:
    print(count)
    time.sleep(1)
    count=count+1

print("Done!")
```

### Behavior

```
>>> %Run -c $EDITOR_CONTENT
start
connect
VfsFat
mount
mount ok
init i2s
filesize of /sd/music/xzl2.wav is 15539848
play wav
Playing wav file!
0
1
PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (ConnectionError: read failed: [Errno 6] Device not configured).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your CircuitPython device and try again.

Process ended with exit code 1.
Unable to connect to /dev/cu.usbmodem0740D1EC05011: [Errno 2] could not open port /dev/cu.usbmodem0740D1EC05011: [Errno 2] No such file or directory: '/dev/cu.usbmodem0740D1EC05011'
```

### Description

Board: adafruit-esp32-s3-tft-feather
SDCard: SPI (MI, MO, SCK, RX-CS), 3V3
Audio: MAX98357A (A0-BCLK, A1-LRC, A2-DIN, A3-GAIN), 3V3
Speaker: 4O 3W
Wav file: /sd/music/xzl2.wav, 16Bits, 22KHz, filesize is 15539848

When I run the code, the speaker plays sound.
[image]

But after a while, the board reboots and enters the boot mode.
[image]

The restart time is uncertain, sometimes after outputting 1 2, it restarts.
Sometimes, the output number exceeds 30 before restarting.
[image]

### Additional information

1. wav files : https://drive.google.com/drive/folders/1adCrrW25AAxZvsPk7WaffKeor43fbk-d?usp=sharing
[image]

2. code to check whether the wav file can be read normally
```
import adafruit_sdcard
import board
import busio
import digitalio
import storage
import math
import os

print("start")
spi = board.SPI()

#print("deinit")
#spi.deinit()

#print("init")
#spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = digitalio.DigitalInOut(board.RX)

print("connect")
sdcard = adafruit_sdcard.SDCard(spi, cs)

print("VfsFat")
vfs = storage.VfsFat(sdcard)

print("mount")
storage.mount(vfs, "/sd")
print("mount ok")

print("write")
with open("/sd/test.txt", "w") as f:
    f.write("Hello world!\r\n")
    f.write("I'am circuitpython!\r\n")

print("read")
with open("/sd/test.txt", "r") as f:
    print("Printing lines in file:")
    for line in f:
        print(line, end='')

#wav_file = "/sd/music/StreetChicken.wav"
wav_file = "/sd/music/xzl2.wav"
stats = os.stat(wav_file)
filesize = stats[6]
print("filesize of %s is %d" % (wav_file, filesize))

times = math.ceil(filesize/1024)
print("times=%d" % times)
    
count = 0
with open(wav_file, "rb") as f:
    while count<times:
        f.seek(count * 1024)
        data = f.read(1000)
        if count % 1000 == 0:
            print("[%d/%d]: " % (count, times), end='')
            print([hex(i) for i in data[0:10]])
        count = count + 1

print("end")

```

[image]

jepler, 2023-07-08:
I wonder if this has any relation to https://github.com/adafruit/circuitpython/issues/8041

HonestQiao, 2023-07-08:
> I wonder if this has any relation to #8041
On adafruit_feather_esp32s3_tft, or on S3 Box Lite, I've tested, but none of the problems described in that article occur.
However, the cause of the problem may be really similar.

In addition, my problem will also appear on 8.2.0.
Adafruit CircuitPython 8.2.0 on 2023-07-05; Adafruit Feather ESP32-S3 TFT with ESP32S3.

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.

closed 2024-01-31 by tannewt