adafruit/circuitpython #8146: Hardware fault in new ESP32-S2 feather

BCPdoc, 2023-07-08:
### CircuitPython version

```python
8.2.0
```

### Code/REPL

```python
# weather station for disabling automatic sprinklers
# https://learn.adafruit.com/weather-wise-wifi-umbrella-stand/code-install-walkthrough
import os
import ssl
import microcontroller
import board
import wifi
import socketpool
import adafruit_requests

if __name__ == '__main__':
    def connect_to_network():
        #  connect to SSID
        wifi.radio.enabled = True
        wifi.radio.connect(os.getenv('CIRCUITPY_WIFI_SSID'), os.getenv('CIRCUITPY_WIFI_PASSWORD'))

        pool = socketpool.SocketPool(wifi.radio)
        requests = adafruit_requests.Session(pool, ssl.create_default_context())

        #pool = socketpool.SocketPool(wifi.radio)
        return requests

    def disconnect_from_network():
        wifi.radio.enabled = False

    # function for making http requests with try/except
    def get_request(tries, ping):
        for i in range(tries):
            try:
                n = ping
            except Exception as error:
                print(error)
                time.sleep(10)
                if i < tries - 1:
                    continue
                raise
            break
        return n

    def check_weather(weather_url):
        requests = connect_to_network()
        response = get_request(5, requests.get(weather_url))
        response_as_json = response.json()
        disconnect_from_network()
        return response_as_json

    def calc_weighted_precip_history(response):
        data = response['daily']['precipitation_sum']
        #weather_url requests 3 historical days, don't care about future
        weights = [ 0.1, 0.1, 0.4, 0.4 ]
        return mult_lists(data,weights)

    def get_precip_prob(response):
        data = response['daily']['precipitation_probability_max']
        #don't care about 3 historical days but only today and tomorrow
        weights = [ 0.0, 0.0, 0.0, 0.7, 0.3]
        return mult_lists(data,weights)

    def mult_lists(data,weights):
        result = 0.0
        for i in range(len(weights)):
            result += weights[i] * data[i]
            #print(str(round(weights[i],2)) + " x " + str(round(data[i],2)) + " = " + str(round(result,2)))
        return result

    def get_current_datetime(response):
        return response['current_weather']['time']

    def interrupt_sprinklers():
        lat = os.getenv("MY_LAT")
        long = os.getenv("MY_LONG")

        # API request to open-meteo
        weather_url = "https://api.open-meteo.com/v1/forecast?"
        # pass latitude and longitude
        weather_url += "latitude=" + lat + "&longitude=" + long + "&daily=precipitation_sum,precipitation_probability_max&current_weather=true&temperature_unit=fahrenheit&windspeed_unit=mph&precipitation_unit=inch&timezone=auto&past_days=3"
        # https://api.open-meteo.com/v1/forecast?latitude=47.718&longitude=-116.9516&hourly=rain&daily=precipitation_sum&timezone=auto&start_date=2023-05-01&end_date=2023-07-05
        # this one is better
        # https://api.open-meteo.com/v1/forecast?latitude=47.718&longitude=-116.9516&daily=precipitation_sum&timezone=auto&past_days=3&forecast_days=3
        # even better
        # https://api.open-meteo.com/v1/forecast?latitude=47.718&longitude=-116.9516&daily=precipitation_sum,precipitation_probability_max&current_weather=true&temperature_unit=fahrenheit&windspeed_unit=mph&precipitation_unit=inch&timezone=auto&past_days=3

        response = check_weather(weather_url)
        #response = {'longitude': -116.956, 'timezone_abbreviation': 'PDT', 'latitude': 47.7152, 'generationtime_ms': 0.645995, 'daily_units': {'time': 'iso8601', 'precipitation_sum': 'inch', 'precipitation_probability_max': '%'}, 'current_weather': {'windspeed': 9.7, 'weathercode': 1, 'is_day': 1, 'time': '2023-07-07T15:00', 'winddirection': 85.0, 'temperature': 90.7}, 'elevation': 665.0, 'daily': {'time': ['2023-07-04', '2023-07-05', '2023-07-06', '2023-07-07', '2023-07-08', '2023-07-09', '2023-07-10', '2023-07-11', '2023-07-12', '2023-07-13'], 'precipitation_sum': [0.1, 0.2, 0.3, 0.0, 0.0, 0.0, 0.048, 0.0, 0.0, 0.0], 'precipitation_probability_max': [0, 0, 13, 0, 0, 3, 48, 42, 0, 3]}, 'timezone': 'America/Los_Angeles', 'utc_offset_seconds': -25200}

        weighted_precip_history = calc_weighted_precip_history(response)
        print("\nPrecipitation history: " + str(round(weighted_precip_history,2)))
        precip_prob = get_precip_prob(response)
        print("Future probability: " + str(round(precip_prob,2)))

        interrupt_sprinklers = False

        if(precip_prob > 95 or weighted_precip_history > 0.1):
            interrupt_sprinklers = True
        else:
            interrupt_sprinklers = False

        return interrupt_sprinklers

    print("Interrupt sprinklers? " + str(interrupt_sprinklers()))
```

### Behavior

Code runs successfully then stops and reports "Fault detected by hardware."

### Description

It worked yesterday but continues to fault after reinstalling from here https://circuitpython.org/board/adafruit_feather_esp32s2_tft/.

### Additional information

none

dhalbert, 2023-07-08:
Please post the full error message, including any line number info.

BCPdoc, 2023-07-08:
Commenting out this line seems to eliminate the fault:

        wifi.radio.enabled = False

BCPdoc, 2023-07-08:
There is no line number. The code finishes running (using serial port if connected), then the board disconnects from the Mu editor / windows, reboots, briefly displays "static", and finally displays "fault detected by hardware".
[Photos-001 (16).zip](https://github.com/adafruit/circuitpython/files/11992532/Photos-001.16.zip)

sveinungkb, 2023-07-24:
This is similar to the issue I'm having with ESP32-S3. Same error message, been scratching my head for a few days now and slimmed down anything wifi/socket related while still using wifi + AP + adafruit httpserver.

DJDevon3, 2023-07-30:
Title says installed on new ESP32-S2 Feather but links at the bottom to the ESP32-S2 TFT UF2.  The ESP32-S2 Feather is different from the ESP32-S2 TFT. When posting the board's boot_out.txt it's best to include everything in it except the UID. 

Very cool script. It's best to include a check for if connected to wifi with a try/except while making the request or ping. If Wifi goes down during the return it might crash the script. I don't think it would crash to safemode though. Checking for ping is good but checking for if wifi.connected during the request process (ping) should still be added somewhere.

Is it possible with your current code to run out of sockets if it thinks wifi is still connected but continues to ping loop over and over? It should break yes. I only mention it because in the past running out of sockets would hard fault and if that issue has somehow returned to show up again in 8.2.1 it's worth a dev looking into.  I believe requests are given a max of 3 sockets so would only have to loop 3 times before running out of sockets.  It "should" throw an exception for that now but in the past it would hard fault.

DJDevon3, 2023-07-30:
Ran his code and was able to replicate the hard fault.
```py
Adafruit CircuitPython 8.2.1 on 2023-07-25; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
```
```py
code.py output:
https://api.open-meteo.com/v1/forecast?latitude=40.7259&longitude=-74.0055&daily=precipitation_sum,precipitation_probability_max&current_weather=true&temperature_unit=fahrenheit&windspeed_unit=mph&precipitation_unit=inch&timezone=auto&past_days=3

Precipitation history: 0.0
Future probability: 36.8
Interrupt sprinklers? False

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
It ran successfully once. said code done running then about 2 seconds later board hard reset and hard faulted to safemode.

```py
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Fault detected by hardware.
Please file an issue with your program at https://github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```
Removed disconnect_from_network() within check_weather function. Ran once and did not hard fault after running.
```py
code.py output:
https://api.open-meteo.com/v1/forecast?latitude=40.7259&longitude=-74.0055&daily=precipitation_sum,precipitation_probability_max&current_weather=true&temperature_unit=fahrenheit&windspeed_unit=mph&precipitation_unit=inch&timezone=auto&past_days=3

Precipitation history: 0.0
Future probability: 36.8
Interrupt sprinklers? False

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

tannewt, 2024-01-31:
I just tried this example on a Feather ESP32-S2 and don't see the crash. Please try with a recent 9.0 beta.

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.

closed 2024-01-31 by tannewt