adafruit/circuitpython #8147: Assignment to wifi.radio.hostname is merged into existing value, not replaced

adrianblakey, 2023-07-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-rc.1 on 2023-06-27; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import wifi
wifi.radio.hostname = 'some-long-value'
print(wifi.radio.hostname) # some-long-value
wifi.radio.hostname = 'a-value'
print(wifi.radio.hostname) # a-valueng-value
```

### Behavior

The value of wifi.radio.hostname is not replaced by the new value, but replaced.

anecdata, 2023-07-08:
Occurs on Pico W, but not on Espressif.

microdev1, 2023-07-09:
The issue is in the following line, `strlen` dosen't count the null terminator.

https://github.com/adafruit/circuitpython/blob/95535a8cd031ffe2780f19f80d5ffdabfb4eecd2/ports/raspberrypi/common-hal/wifi/Radio.c#L91

closed 2023-07-09 by dhalbert