adafruit/circuitpython #8149: Out of MDNS service slots

adrianblakey, 2023-07-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0 on 2023-07-05; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import wifi
import mdns
import socketpool
from adafruit_httpserver import Server

# set SSID and pwd in settings.toml

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, debug=True)
try:
    mdns_server = mdns.Server(wifi.radio)
    mdns_server.hostname = wifi.radio.hostname
    mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=80)
except RuntimeError as e:
    print("MDNS setting failed:", str(e))
```

### Behavior

MDNS setting failed: Out of MDNS service slots  

Please explain what causes this and how to avoid/fix it.

anecdata, 2023-07-09:
```py
import wifi
import mdns

try:
    mdns_server = mdns.Server(wifi.radio)
    mdns_server.hostname = wifi.radio.hostname
    mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=80)
except RuntimeError as e:
    print("MDNS setting failed:", str(e))
```

Works if web workflow is not enabled in `settings.toml`, but not if web workflow is enabled.

mDNS only allows one service_type and protocol per port, and web workflow automatically sets up an HTTP (TCP) service on port 80 (port is configurable in `settings.toml`). With web workflow enabled, a different port can be used. For example:
```py
import wifi
import mdns

try:
    mdns_server = mdns.Server(wifi.radio)
    mdns_server.hostname = wifi.radio.hostname
    mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=8080)
except RuntimeError as e:
    print("MDNS setting failed:", str(e))
```

adrianblakey, 2023-07-09:
ty

dhalbert, 2023-07-09:
@anecdata Do you think this should be explained in the doc?

anecdata, 2023-07-09:
It's mentioned in the `mdns` [readthedocs](https://docs.circuitpython.org/en/latest/shared-bindings/mdns/index.html#mdns.Server.advertise_service), but not the complication of `mdns` + web [workflow](https://docs.circuitpython.org/en/latest/docs/workflows.html#web).

anecdata, 2023-07-10:
I've been accumulating a short list of readthedocs edits to PR for port-dependent network limitations, I'll add this.

closed 2023-07-17 by tannewt

adrianblakey, 2023-07-18:
Another observation which you might want to document ...

I have web workflow enabled and I run this:

```
import wifi
import mdns

mdns_server = mdns.Server(wifi.radio)
mdns_server.hostname = wifi.radio.hostname # 'logger'
mdns_server.instance_name = "My service"
```

I then run the Discovery app on Mac OS to browse the bonjour advertisements. There is no indication that my Pi provides a _http service - the following is displayed though:

```
...
_circuitpython._tcp.
> My Service
   > logger.local.
      192.168.178.89:80
...
```

anecdata, 2023-07-18:
I'm not sure what's best here for docs. There isn't a new service being advertised, but hostname and instance name are changed. Web Workflow is documented as `_circuitpython._tcp.` <https://docs.circuitpython.org/en/latest/docs/workflows.html#web>

anecdata, 2024-04-01:
Footnote: it does not appear possible to have the same MDNS service-protocol on different ports:
> service_type and protocol can only occur on one port. Any call after the first will update the entry’s port.

A non-MDNS HTTP Server can be run in code.py with web workflow enabled, but not an MDNS HTTP Server. User code could also (temporarily to permanently) commandeer the web workflow  _http._tcp service-protocol to its own port for an in-`code.py` MDNS HTTP Server, but at that point, web workflow will no longer operate.

https://docs.circuitpython.org/en/latest/shared-bindings/mdns/index.html#mdns.Server.advertise_service