adafruit/circuitpython #8151: mimxrt10xx: Implement canio (micromod)

qutefox, 2023-07-09:
I have started working on a canio implementation for the Teensy Micromod (mimxrt10xx port).
I am not a circuitpython nor a CAN bus expert but I hope I can make a working version and pass on enhancing and supporting/testing more mimxrt10xx boards (other than Micromod) to others.

I am making this git issue to track/publicize my work (according to this forum reply: https://forums.adafruit.com/viewtopic.php?p=978568#p978568)

If there is any CAN bus expert who could do code review or make suggestions please let me know.

tannewt, 2023-07-10:
Thank you! We're happy to review when you have some code you are ready to share. It doesn't need to be perfect. :-)

qutefox, 2023-07-28:
I made the neccessary changes however when I try to test using the following setup:
Teensy MicroMod board [RX, TX pins] <==> [RX, TX pins] TI SN65HVD230 CAN-bus transceiver module (3.3V) [CAN H, L pins] <==> [CAN H, L pins] MCP2515 CAN Bus Controller Module [SPI bus] <==> 5v to 3.3V logic level shifter <==> Adafruit QT Py
ESP32-S3 No PSRAM
it doesn't work with the following code examples:

MCP2525 side.
`from time import sleep
import board
from digitalio import DigitalInOut
from adafruit_mcp2515.canio import Message, RemoteTransmissionRequest
from adafruit_mcp2515 import MCP2515 as CAN

cs = DigitalInOut(board.A0)
cs.switch_to_output()

can_bus = CAN(board.SPI(), cs, baudrate=250000)
message = Message(id=0x0408, data=b"adafruit")

can_bus.send(message)

while True:

    with can_bus.listen(timeout=1.0) as listener:
        message_count = listener.in_waiting()
        if message_count > 0:
            print(f"Messages available: {message_count}")
            for _i in range(message_count):
                msg = listener.receive()
                print("Message from ", hex(msg.id))
                if isinstance(msg, Message):
                    print("message data:", msg.data)
                if isinstance(msg, RemoteTransmissionRequest):
                    print("RTR length:", msg.length)
    sleep(1)
`

Teesy MicoMod side.
`from time import sleep
import canio
import board

can = canio.CAN(board.MM_CAN_TX, board.MM_CAN_RX, baudrate=250000)
message = canio.Message(id=0x0408, data=b"adafruit")

can.send(message)

while True:
    with can.listen(timeout=1.0) as listener:
        message_count = listener.in_waiting()
        if message_count > 0:
            print(f"Messages available: {message_count}")
            for _i in range(message_count):
                msg = listener.receive()
                print("Message from ", hex(msg.id))
                if isinstance(msg, Message):
                    print("message data:", msg.data)
                if isinstance(msg, RemoteTransmissionRequest):
                    print("RTR length:", msg.length)
    sleep(1)
`

Using a logic analyser I can see that the Teensy sends out the correct signals on the TX pin, however the SN65HVD230 fails to put anything on the CAN H-L lines (set logic level threshold to 2.5V).
When the MCP2515 sends I can see the signal on the CAN H-L lines but nothing on the RX pin after the SN65HVD230  chip.

Tried two SN65HVD230 chips both gave me the same result. I am not sure what to do. I think I'll leave my git patch file here for other to pick up and possibly test.

[image]

[image]

[canio_mimxrt10xx_git.patch](https://github.com/adafruit/circuitpython/files/12192407/canio_mimxrt10xx_git.patch)

qutefox, 2023-07-28:
I've tested my MCP2515 board with VESC controllers and I can receive and send without any issues. So the problem must not be on that side. I am 99% sure that the SN65HVD230 chips must be bad. I read online that there are a lot of fake chips. I don't know where to get a genuine one. Maybe I'll try to order one from Farnell or DigiKey.

tannewt, 2023-07-28:
Would this work? https://www.adafruit.com/product/5708

closed 2025-01-21 by tannewt