adafruit/circuitpython #8157: Pi Pico WiFi scanning does not list an iPhone personal hotspot

adrianblakey, 2023-07-11:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0 on 2023-07-05; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import wifi

for network in wifi.radio.start_scanning_networks():
    str(network.ssid, "utf-8"), network.rssi, network.channel))
wifi.radio.stop_scanning_networks()
```

### Behavior

Does not detect an iPhone personal hotspot - set with "Maximize Compatibility"

Is this a function of the WiFi chip or the hardware or an Apple issue. Needless to say the ssid of the ph is discovered by other devices.

anecdata, 2023-07-11:
It's _possible_ the phone is creating a hotspot only at 5 GHz. Pico W (and current Espressif ESP32* chips) support only 2.4 GHz.

adrianblakey, 2023-07-16:
I think I know the reason. The personal hotspot SSID is only broadcast while the personal hotspot screen is open on the iPhone and (perhaps) for a short period of time after it's closed. I have successfully connected under these conditions.

closed 2023-07-16 by adrianblakey