adafruit/circuitpython #8160: raspberrypi: support sdioio

jepler, 2023-07-11:
There is appropriately-licenesed code in pico-extras to support SD cards in SDIO mode. Adapt this code for CircuitPython.

DatanoiseTV, 2024-10-09:
I recommend to look at the repo I have referenced. The SDIO is much faster than the one in pico-extra.

dhalbert, 2024-10-09:
From #9702 (dupe):

@DatanoiseTV 
> As part of work on the PicoADK V2 and initial CircuitPython support, I was wondering if SDIO support is coming?

> This works really well https://github.com/earlephilhower/arduino-pico/discussions/1560#discussioncomment-10681395

DatanoiseTV, 2024-10-21:
> From #9702 (dupe):
> 
> @DatanoiseTV
> 
> > As part of work on the PicoADK V2 and initial CircuitPython support, I was wondering if SDIO support is coming?
> 
> > This works really well [earlephilhower/arduino-pico#1560 (comment)](https://github.com/earlephilhower/arduino-pico/discussions/1560#discussioncomment-10681395)

Would be great to see support for this in 9.2.

jepler, 2024-10-22:
Adafruit has categorized this enhancement as "long term", meaning that Adafruit folks are not likely to be working on it in the near future. We would welcome community members picking up the work, of course.

DatanoiseTV, 2024-12-12:
We are soon releasing the PicoADK v2 https://circuitpython.org/board/datanoise_picoadk_v2/ with an SD attached via SDIO (with UHS support). Would be great to also have support for it in CircuitPython. The main SDK so far is Arduino and a bare C.