adafruit/circuitpython #8172: minimqtt loop exception when using TLS

berarduc, 2023-07-14:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-19-g8de9d5a52 on 2023-07-13; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import os
#import ipaddress
import wifi
import socketpool
import ssl
#import adafruit_requests
import time
import board
import digitalio
import adafruit_minimqtt.adafruit_minimqtt as MQTT
import microcontroller

WIFI_SSID = "mySSID"
WIFI_PW = "myPW"

MQTT_SUB_TOPIC = "mySubTopic"
MQTT_PUB_TOPIC = "myPubTopic"
HA_MQTT_BROKER = "myBrokerIpAdd"

def restart(mqttClient):
    print("\n\nResetting in 10 sec...\n\n")
    time.sleep(1)
    mqttClient.disconnect()
    time.sleep(1)
    mqttClient.reconnect()
    time.sleep(1)

#
#	Set Up MQTT
#

# Define callback methods which are called when events occur
# pylint: disable=unused-argument, redefined-outer-name
def connect(mqtt_client, userdata, flags, rc):
    # This function will be called when the mqtt_client is connected
    # successfully to the broker.
    print("Connected to MQTT Broker!")
    print("Flags: {0}\n RC: {1}".format(flags, rc))

def disconnect(mqtt_client, userdata, rc):
    # This method is called when the mqtt_client disconnects
    # from the broker.
    print("Disconnected from MQTT Broker!")

def subscribe(mqtt_client, userdata, topic, granted_qos):
    # This method is called when the mqtt_client subscribes to a new feed.
    print("Subscribed to {0} with QOS level {1}".format(topic, granted_qos))

def unsubscribe(mqtt_client, userdata, topic, pid):
    # This method is called when the mqtt_client unsubscribes from a feed.
    print("Unsubscribed from {0} with PID {1}".format(topic, pid))

def publish(mqtt_client, userdata, topic, pid):
    # This method is called when the mqtt_client publishes data to a feed.
    print("Published to {0} with PID {1}".format(topic, pid))

def message(client, topic, message):
    print("New message on topic {0}: {1}".format(topic, message))

def setUpMqtt(pool):

        sslContext = ssl.create_default_context()
        
        # read cert into memory
        
        with open('ca-root-cert.crt','r') as f:
            fileData = f.read()
            f.close
        
        sslContext.load_verify_locations(cadata=fileData)
        
        # Set up a MiniMQTT Client

        mqtt_client = MQTT.MQTT(
            broker=HA_MQTT_BROKER,
            username="username",
            password="password",
            port=8883,
            is_ssl=True,
            socket_pool=pool,
            ssl_context = sslContext,
         )
        
        # Connect callback handlers to mqtt_client
        mqtt_client.on_connect = connect
        mqtt_client.on_disconnect = disconnect
        mqtt_client.on_subscribe = subscribe
        mqtt_client.on_unsubscribe = unsubscribe
        mqtt_client.on_publish = publish
        mqtt_client.on_message = message

        print("Attempting to connect to %s" % mqtt_client.broker)
        mqtt_client.connect()
        
        return mqtt_client

# NOTE watchdog not used currently
def connectToWifi(watchdog):
    print()
    print("Connecting to WiFi...")
    if watchdog != None:
        wd_feed(watchdog)
    wifi.radio.connect(WIFI_SSID,WIFI_PW)
    if watchdog != None:
        wd_feed(watchdog)
    
    pool = socketpool.SocketPool(wifi.radio)

    print("My MAC addr: ",[hex(i) for i in wifi.radio.mac_address])

    print("My IP address: ",wifi.radio.ipv4_address)

    return pool

def main():
    print("\n\nmqttTest Starting up...\n\n")
    
    try:
        pool = connectToWifi(None)
    except:
        print("\n\n---> ERROR Cannot Connect to WiFi.\n\n")
    else:
    
        print("Connected to WiFi.\n\n")
        print("Setting up MQTT...")
        client = setUpMqtt(pool)
        client.publish(MQTT_PUB_TOPIC,"hello from pico")
        print("Subscribing to %s" % MQTT_SUB_TOPIC)
        client.subscribe(MQTT_SUB_TOPIC)
        while True:
            try:
                response = client.loop()
            except:
                print("\n\n---> ERROR in Mqtt Loop \n\n")
                if client.is_connected():
                    print("CLIENT STILL CONNECTED")
                else:
                    print("CLIENT NOT CONNECTED")
                restart(client)
                client.subscribe(MQTT_SUB_TOPIC)
            else:
                if response != None:
                    print("client loop response = ",response)
                time.sleep(1)
        print("Disconnecting from %s" % client.broker)
        client.disconnect()
        print("\n\nmqttTest Shutting down...\n\n")
    
main()
```

### Behavior

pico successfully connects to WiFi, and then pico successfully connects to mqtt broker without fail.

pico then subscribes to mqtt topic, but when LOOP() is executed, it ALWAYS throws an exception.

NOTE: If TLS is disabled, then it ALWAYS works.

### Description

This behavior seems similar to issue #7606 which has been closed. 

I'm using circuitpython 8.2 with associated minimqtt library.

Any help would be appreciated.

Thanks.

anecdata, 2023-07-14:
Can you post the full exception trace?

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.

closed 2024-01-31 by tannewt