adafruit/circuitpython #8173: OrderedDict.fromkeys returns an (unordered) dict

kbsriram, 2023-07-14:
### CircuitPython version

```python
Adafruit CircuitPython 8.1.0 on 2023-05-22; DFRobot Beetle ESP32-C3 with ESP32-C3FN4
```

### Code/REPL

```python
>>> from collections import OrderedDict
>>> d = OrderedDict.fromkeys('abcdefg')
>>> ''.join(d)
'agcbdef'
>>> type(d)
<class 'dict'>
>>>
```

### Behavior

- The returned type should be `<class 'OrderedDict'>` - instead of `<class 'dict'>`
- The object should return sorted keys (`'abcdefg'`), instead it is in unsorted `dict` order (`'agcbdef'`)

### Additional information

The `fromkeys` classmethod directly uses `mp_obj_new_dict` https://github.com/adafruit/circuitpython/blob/8de9d5a52259cd0422a3633b6cf562e44d0cca9d/py/objdict.c#L278-L284  but misses  the OrderedDict related initializations used for the constructor https://github.com/adafruit/circuitpython/blob/8de9d5a52259cd0422a3633b6cf562e44d0cca9d/py/objdict.c#L116-L121

Might it be simplest move the shared initialization into a function that's callable from the constructor and `fromkeys`? (Lmk if you folks don't have bandwidth to fix any time soon, but have cycles to review a PR :smile: )

tannewt, 2023-07-14:
> (Lmk if you folks don't have bandwidth to fix any time soon, but have cycles to review a PR smile )

We're totally game to review any PRs you feel like making! Thanks!

Also, check the MicroPython implementation first because it may have already been fixed there. We're currently working to update from 1.18 to (future) 1.21 for CP 9.0.

dhalbert, 2023-07-14:
This may well be a problem upstream in micropython as well: https://github.com/micropython/micropython/blob/master/py/objdict.c. There has been some discussion about the new Python requirement that dictionaries are naturally ordered.

closed 2023-07-17 by tannewt