adafruit/circuitpython #8193: Circuitpython 8.2 + esp32-s2-wroom board pin problem

suikula, 2023-07-22:
Hello, 

I have tested with stable 8.2 (https://circuitpython.org/board/espressif_saola_1_wroom/) and with adafruit-circuitpython-espressif_saola_1_wroom-en_US-20230721-823e505.bin. 

I have esp32-s2-wroom board and while trying to use DAC pins Im not able to. Module board wont show correct pins.

```
import board
dir(board)
['__class__', '__name__', 'IO0', 'IO1', 'IO10', 'IO11', 'IO12', 'IO13', 'IO14', 'IO15', 'IO16', 'IO17', 'IO18', 'IO19', 'IO2', 'IO20', 'IO21', 'IO26', 'IO3', 'IO33', 'IO34', 'IO35', 'IO36', 'IO37', 'IO38', 'IO39', 'IO4', 'IO40', 'IO41', 'IO42', 'IO43', 'IO44', 'IO45', 'IO46', 'IO5', 'IO6', 'IO7', 'IO8', 'IO9', 'NEOPIXEL', 'RX', 'TX', 'board_id']
```

[https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/_images/esp32-s2_saola1-pinout.jpg](url)

There is no A0 or A1.

Im using ubuntu 20.04 and Mu editor.

Am I missing something? Any tips?

dhalbert, 2023-07-22:
Yes, use the `IOnn` pins. Are you using an Espressif dev board or some other board? Post a shopping link to the board you used if it is not the Espressif board so we can see a picture.

suikula, 2023-07-22:
> Pin names like `A0` or `A1` are convenience aliases that are usually added when the board silk has those labels. On the ESP chips, there are two banks of ADCs - generally you want to use `ADC1_*` pins (`IO01` - `IO10`) if also using wifi. There are two DAC pins: `IO17` and `IO18`, but `IO18` is used for the NeoPixel.

Thx for the reply.  Using IO17 atleast got me forward but other python code problems prevents me from actually confirming that this is working.

suikula, 2023-07-22:
> Yes, use the `IOnn` pins. Are you using an Espressif dev board or some other board? Post a shopping link to the board you used if it is not the Espressif board so we can see a picture.

Yep IOnn pins most likely works.
https://www.ebay.com/itm/385252950700?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=ntQFEp1PQuy&sssrc=4429486&ssuid=C6bd3pbsS-m&var=653138665549&widget_ver=artemis&media=COPY

This is the board I am using.

dhalbert, 2023-07-22:
OK, yes, this is a Saola clone, so use the IO pins as described. If you are having other issues you can't figure out, feel free to ask in our discord https://adafru.it/discord or in the forums: https://forums.adafruit.com/viewforum.php?f=60.

closed 2023-07-22 by dhalbert