adafruit/circuitpython #8232: [Request for coments] TOML parsing library

elpekenin, 2023-07-29:
I've been working on a (still very incomplete) library for parsing TOML, see it [here](https://github.com/elpekenin/circuitpython_toml).
Sharing it in for visibility, just in case someone wants to use it or contribute, or core team wants to merge it (after some fixes).

Supported types:
- ints
- floats
- bin/oct/hex literals
- strings
- lists (details below)

Current workflow is
- For each line in the input:
  1. Iterate the line to find some metadata, eg tokens (like quotes or brackets) found
  2. Run some basic syntax checking on these (eg: you need some content after `=`)
  3. If line is valid, evaluate it:
      - On lines setting the scope (`[scope]`), remember it
      - When a line is empty, reset this state
      - If there's an assignment (`key = value`), try and cast value to supported ones (unless it is quoted), leave as string otherwise.

Made a couple questionable decissions along the way:
- Allowing unquoted strings, it's friendlier (in my opinion), but on "real" TOML they would be invalid
```toml
# both of these are treated the same by my code
foo = "bar"
foo = bar
```
(see how syntax hightlighter doesnt like it)
- `True` and `False` also qualify as booleans (to conform Python's syntax), despite TOML having these literals as all-lowercase - in fact, the parser is case-insensitive at the moment
```toml
# all of these are also the same
foo = true
foo = True
foo = trUE
```

Broken behaviour (non-exhaustive list):
- Triple quotes (ie: multiline comments)
- Elements inside a list can't be strings nor other lists

---

Could fix #4303, on a quick glance, it seems like `ini` is pretty much `toml` without requiring quotes eiter

Could improve `os.getenv` behaviour, which currently supports only ints and strings (booleans to control behaviour of code look better than using 0/1, imo)

I know bill88t (GitHub wont show the username for @-tagging+ i dont want to ping) has also been working on one, and we have discussed a bit on the topic, but we have very different approaches, so here is mine.

tannewt, 2023-07-31:
I'd prefer to be strictly TOML. By varying, you are leaving the potential for non-portable code.

elpekenin, 2023-07-31:
> strict TOML

Ok, noted, thanks for the feedback.

Will try and get it pushed tomorrow, any other ideas so far?

elpekenin, 2023-08-03:
First and foremost... the code i pushed the other day was an old version by mistake. I pushed the newer one yesterday.

I have now pushed:
- Require quotes for things to be considered strings
- bools have to be all-lowercase
- Implemented better parsing of lists, it somewhat works but breaks/parses wrong on some scenarios. Will try and fix it during the week, but a second pair of eyes would definitely help locating bugs :)