adafruit/circuitpython #8237: Fault detected by hardware when using socketpool

superhac, 2023-07-30:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-45-g75ba17e84 on 2023-07-28; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
Board ID:espressif_esp32s3_devkitc_1_n8r2
```

### Code/REPL

```python
import board
import adafruit_rgbled
import neopixel
import wifi
import os
import socketpool
import ipaddress
import microcontroller

def processNetMsgs():
    try:
        buff = bytearray(128)
        numbytes, addr = sock.recvfrom_into(buff)
        if numbytes:
            msg = buff.decode('utf-8')
            print(f"Received message from {addr[0]} / {numbytes}: ", msg)
    except OSError as e:
		#print(e)
		pass

# setup net 192.168.99.X
wifi.radio.set_ipv4_address_ap(ipv4=ipaddress.IPv4Address("192.168.99.1"), netmask=ipaddress.IPv4Address("255.255.255.0"),gateway=ipaddress.IPv4Address("192.168.99.1"))
wifi.radio.start_ap(ssid=os.getenv("CIRCUITPY_WIFI_SSID"),password=os.getenv("CIRCUITPY_WIFI_PASSWORD"))
print("WiFi SSID:",os.getenv("CIRCUITPY_WIFI_SSID"))
print("MAC addr:", [hex(i) for i in wifi.radio.mac_address])
print("AP IP addr:", wifi.radio.ipv4_address_ap)

# setup UDP Broadcast listener
pool = socketpool.SocketPool(wifi.radio)
udp_port = 9797
sock = pool.socket(pool.AF_INET, pool.SOCK_DGRAM) # UDP socket
sock.bind(("0.0.0.0", udp_port))  # broadcastr bind
#sock.settimeout(None)
sock.setblocking(None)
sock.listen(10)

# setup neopixel to represent good boot up
pixelLed = neopixel.NeoPixel(board.NEOPIXEL, 1)
pixelLed.brightness = .01
pixelLedOn = True
pixelLed.fill((0, 255, 0)) # Great LED is a go!

# main loop
while True:
    processNetMsgs()
```

### Behavior

```You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Fault detected by hardware.
Please file an issue with your program at https://github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.```

### Description

Anywhere between 5 and 15 minutes of running the esp will enter safemode with the "Fault detected by hardware" runtime error.  From what I observed it appears to be trigger from the `recvfrom_into` function in the socket pool library.  The only thing in this code that i could imagine to be unique is the fact its non-blocking udp and its listening for broadcast traffic.

DJDevon3, 2023-07-30:
```py
Adafruit CircuitPython 8.2.1 on 2023-07-25; Adafruit Feather ESP32S2 with ESP32S2
Board ID:adafruit_feather_esp32s2
```
Setup AP and was able to connect to it with my phone. Green light on, everything looks good.
```py
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
WiFi SSID: CustomNet
MAC addr: [obfuscated]
AP IP addr: 192.168.3.192
Time : 1843.76
```
After a while decided to soft reboot to see if that in itself would cause a hard fault. 
```py
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
WiFi SSID: CustomNet
MAC addr: [obfuscated]
AP IP addr: 192.168.3.192
Time : 4381.54
```
Runtime approximately 42 minutes without hard fault. Was unable to replicate this issue.

anecdata, 2023-07-31:
After some discussions on Discord, another possibility to rule out:

Having `CIRCUITPY_WIFI_SSID` and `CIRCUITPY_WIFI_PASSWORD` in `settings.toml` will enable web workflow with auto-connect. That makes this device a wifi station trying to connect to the same SSID and password that this device's Access Point is using. That's probably an unnecessary complication. Separate `settings.toml` settings could be used for the AP credentials, and if web workflow isn't needed, those credentials could be commented out.

I'l try the above code on the same board with the same firmware, without the LED code and unneeded imports and report back.

superhac, 2023-07-31:
So out of desperation I just tried commenting out the line that sets the static IP for operating in AP mode and it did not enter safemode.  It ran all night (12 hours) with out a hiccup.  The updated code is below that worked.  I did take out more things that were not related.  So its a more condensed version now.

```
import board
import wifi
import os
import socketpool
import ipaddress

def processNetMsgs():
    try:
        buff = bytearray(128)
        numbytes, addr = sock.recvfrom_into(buff)
        if numbytes:
            msg = buff.decode('utf-8')
            print(f"Received message from {addr[0]} / {numbytes}: ", msg)
    except OSError as exc:
        if exc.args[0] == errno.EAGAIN:
            pass
        else:
            raise

# setup net 192.168.99.X
wifi.radio.start_ap(ssid=os.getenv("CIRCUITPY_WIFI_SSID"),password=os.getenv("CIRCUITPY_WIFI_PASSWORD"))
#wifi.radio.set_ipv4_address_ap(ipv4=ipaddress.IPv4Address("192.168.99.1"), netmask=ipaddress.IPv4Address("255.255.255.0"),gateway=ipaddress.IPv4Address("192.168.99.1"))

print("WiFi SSID:",os.getenv("CIRCUITPY_WIFI_SSID"))
print("MAC addr:", [hex(i) for i in wifi.radio.mac_address])
print("AP IP addr:", wifi.radio.ipv4_address_ap)

# setup UDP Broadcast listener
pool = socketpool.SocketPool(wifi.radio)
udp_port = 9797
sock = pool.socket(pool.AF_INET, pool.SOCK_DGRAM) # UDP socket
sock.bind(("", udp_port))  # broadcastr bind
sock.setblocking(False)

# main loop
while True:
    processNetMsgs()
```

So now I would guess its not related to the `socketpool`?

superhac, 2023-07-31:
> 

I don't have a safemode.py but I do have the following boot.py:

```
import usb_hid

# This is only one example of a gamepad descriptor, and may not suit your needs.
GAMEPAD_REPORT_DESCRIPTOR = bytes((
    0x05, 0x01,  # Usage Page (Generic Desktop Ctrls)
    0x09, 0x05,  # Usage (Game Pad)
    0xA1, 0x01,  # Collection (Application)
    0x85, 0x04,  #   Report ID (4)
    0x05, 0x09,  #   Usage Page (Button)
    0x19, 0x01,  #   Usage Minimum (Button 1)
    0x29, 0x20,  #   Usage Maximum (Button 32)
    0x15, 0x00,  #   Logical Minimum (0)
    0x25, 0x01,  #   Logical Maximum (1)
    0x75, 0x01,  #   Report Size (1)
    0x95, 0x20,  #   Report Count (32)
    0x81, 0x02,  #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x05, 0x01,  #   Usage Page (Generic Desktop Ctrls)
    0x15, 0x81,  #   Logical Minimum (-127)
    0x25, 0x7F,  #   Logical Maximum (127)
    0x09, 0x30,  #   Usage (X)
    0x09, 0x31,  #   Usage (Y)
    0x09, 0x32,  #   Usage (Z)
    0x09, 0x35,  #   Usage (Rz)
    0x75, 0x08,  #   Report Size (8)
    0x95, 0x04,  #   Report Count (4)
    0x81, 0x02,  #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,        # End Collection
))

gamepad = usb_hid.Device(
    report_descriptor=GAMEPAD_REPORT_DESCRIPTOR,
    usage_page=0x01,           # Generic Desktop Control
    usage=0x05,                # Gamepad
    report_ids=(4,),           # Descriptor uses report ID 4.
    in_report_lengths=(8,),    # This gamepad sends 6 bytes in its report.
    out_report_lengths=(0,),   # It does not receive any reports.
)

usb_hid.enable(
    (usb_hid.Device.KEYBOARD,
     usb_hid.Device.MOUSE,
     usb_hid.Device.CONSUMER_CONTROL,
     gamepad)
)

```

superhac, 2023-08-01:
Alright, I was able to compile and get a debug build running.  But the backtrace gets corrupt each time after multiple attempts.  I will trying in hopes I can get one that doesn't.   

Here's the debug output:
```
FHFuRiBc)Ri~cI (215991) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1
I (215991) wifi:station: 58:1c:f8:a7:d7:e4 join, AID=1, bgn, 40U
I (216021) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.99.2
I (217031) wifi:<ba-add>idx:2 (ifx:1, 58:1c:f8:a7:d7:e4), tid:0, ssn:22, winSize:64
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42013659  PS      : 0x00060e30  A0      : 0x3fcac240  A1      : 0x3fcac220  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x82013659  A9      : 0x3fcf6bb0  
A10     : 0x00000000  A11     : 0x3de0031c  A12     : 0x3de00328  A13     : 0x3de00328  
A14     : 0x3de002fc  A15     : 0x4201cdf5  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x42013656:0x3fcac220 0x3fcac23d:0x3fcac240 |<-CORRUPTED

ELF file SHA256: 1c1618a7e889f689

CPU halted.

```
And I also ran the `decode_backtrace.py` program:
```
? Backtrace: 0x42013656:0x3fcac220 0x3fcac23d:0x3fcac240 |<-CORRUPTED
0x42013656: fun_bc_call at /root/circuitpython/ports/espressif/../../py/objfun.c:297 (discriminator 4)
0x3fcac23d: ?? ??:0
0x00000000: ?? ??:0
```

I;m not sure what to make of that output but maybe someone here does.

sveinungkb, 2023-08-13:
@tannewt you added the 9.0.0 milestone, is there anything specific planned for this?
I'm having a similar issue and eager to validate any fix.

tannewt, 2023-08-14:
We're punting most ESP bugs to 9.0.0 because we are updating to IDF 5.0+. Once we do that update, we'll ask for a retest of this issue and go from there.

superhac, 2023-08-14:
I thought I update this now that I appear to have a work around.  By increasing the bytearray to 65507 (maximum udp payload size) for holding the udp payload I no longer get the hardware failure fault.  But instead I get a `Internal watchdog timer expired` fault anywhere between 17 and 31 hours.   I can work around that by using the `safemode.py` file to reset the esp32.   Now that number maybe arbitrary.  e.g maybe less would work also, but I just stuck with it since I can around the hardware failure fault. 

So the change that works for me is this from the code above:
`buff =  bytearray(65507)`

For now this hackery works for me to get around it.

sveinungkb, 2023-08-19:
I'm not using sockets directly, just the webserver from adafruit (`adafruit_httpserver`). 
Any suggestions what a similar fix would be for http?

superhac, 2023-08-19:
> I'm not using sockets directly, just the webserver from adafruit (`adafruit_httpserver`). Any suggestions what a similar fix would be for http?
> 
> Code is based on https://learn.adafruit.com/pico-w-http-server-with-circuitpython/code-the-pico-w-http-server

Looks like you can change it.  The default is 1024. See https://docs.circuitpython.org/projects/httpserver/en/latest/api.html#adafruit_httpserver.server.Server.request_buffer_size

So in your code you would have to add the request_buffer_size method after instantiating the "Server" object. 
```
print("Connected to WiFi")
pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static", debug=True)
server.request_buffer_size = 65535  # <--------------- Add this line.  Max TCP size
```

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.

closed 2024-01-31 by tannewt