adafruit/circuitpython #8238: MDNS requires a dash in hostname?

ThomasJRyan, 2023-07-31:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.1 on 2023-07-25; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import mdns
import socketpool
import wifi

from adafruit_httpserver import Server, Request, FileResponse

mdns_server = mdns.Server(wifi.radio)
mdns_server.hostname = "helloworld"
mdns_server.advertise_service(service_type="_http", protocol="_tcp", port=80)

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static", debug=True)

wifi.radio.connect(
                ssid="<redacted>",
                password="<redacted>",
            )

@server.route("/")
def base(request: Request):
    """
    Serve the default index.html file.
    """
    return FileResponse(request, "index.html", "/www")

server.serve_forever(str(wifi.radio.ipv4_address))
```

### Behavior

No error messages. 

I am unable to connect to http://helloworld.local

### Description

When I set the hostname to something without a dash such as `helloworld` MDNS does not seem to work correctly and I am unable to connect to http://helloworld.local

However, when I change it to something like `hello-world` then suddenly I am now able to connect to http://hello-world.local

anecdata, 2023-07-31:
Is web workflow enabled via a `settings.toml` file?

Does it make a difference if `wifi.radio.connect` is moved before `mdns_server.advertise_service`?

anecdata, 2023-07-31:
Cool,  as long as `CIRCUITPY_WIFI_SSID` / `CIRCUITPY_WIFI_SSID` aren't in `settings.toml`, there won't be a conflict of two MDNS servers using `_http._tcp.`.