adafruit/circuitpython #8244: SPI frequency being reset by sdcardio

jedgarpark, 2023-08-02:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-beta.0-9-ge631350b9 on 2023-06-06; Metro MIMXRT1011 with IMXRT1011DAE5A
Board ID:metro_m7_1011
```

### Code/REPL

```python
import board, busio, sdcardio
spi = busio.SPI(board.SCK, MISO=board.MISO, MOSI=board.MOSI)
print("before config:", spi. frequency)
while not spi.try_lock():
    pass
spi.configure(baudrate=30_000_000) # max according to mimxrt10xx/common-hal/busio/SPI.c
spi.unlock()
print("after config:", spi.frequency)
cs = board.D7 
sdcard = sdcardio.SDCard(spi, cs, baudrate=30_000_000)
print("after SD card:", spi.frequency)
```

### Behavior

```code.py output:
before config: 249999
after config: 26666666
after SD card: 249999```

### Description

It seems that creating the sdcardio SDCard object brings the SPI bus speed back down to default if I'm understanding this properly. Wav file read performance is very slow using defaults, so trying to speed it up. The behavior seems to be the same if I omit the baudrate call from the sdcard creation.

jedgarpark, 2023-08-04:
@tannewt ah, ok, after doing a read it is 26666666

tannewt, 2023-11-28:
This is by design and not a bug. configure only applies until the next configure is called. It doesn't get undone automatically.

closed 2023-11-28 by tannewt