adafruit/circuitpython #8263: PyPortal Titano not honoring RunMode.UF2 flag on_next_reset

bcr, 2023-08-06:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.1 on 2023-07-25; Adafruit PyPortal Titano with samd51j20
```

### Code/REPL

```python
>>> import microcontroller
>>> microcontroller.on_next_reset(microcontroller.RunMode.UF2)
>>> microcontroller.reset()
```

### Behavior

Reboots to normal running CircuitPython

### Description

When I run the provided code, it just goes back to normal user mode.

### Additional information

If I use the older `RunMode.BOOTLOADER` it mounts the PORTALBOOT drive which is what I want.

I think the problem is here https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/microcontroller/__init__.c#L66

My suggested change is:
```diff
diff --git a/ports/atmel-samd/common-hal/microcontroller/__init__.c b/ports/atmel-samd/common-hal/microcontroller/__init__.c
index d1c02b360..054d7ee7f 100644
--- a/ports/atmel-samd/common-hal/microcontroller/__init__.c
+++ b/ports/atmel-samd/common-hal/microcontroller/__init__.c
@@ -63,7 +63,7 @@ void common_hal_mcu_enable_interrupts(void) {
 }
 
 void common_hal_mcu_on_next_reset(mcu_runmode_t runmode) {
-    if (runmode == RUNMODE_BOOTLOADER) {
+    if ((runmode == RUNMODE_BOOTLOADER) || (runmode == RUNMODE_UF2)) {
         if (!bootloader_available()) {
             mp_raise_ValueError(translate("Cannot reset into bootloader because no bootloader is present"));
         }
```

But this suggestion could be uninformed, dangerous, or just plain wrong. I don't know if there are some SAMD that have both a bootloader and a UF2 mode for instance.

bcr, 2023-08-06:
I was able to build and install the `firmware.uf2` and this patch acts as I expect on the PyPortal Titano.

```
Adafruit CircuitPython 8.2.0-68-ge7e6fc844 on 2023-08-06; Adafruit PyPortal Titano with samd51j20
>>> import microcontroller
>>> microcontroller.on_next_reset(microcontroller.RunMode.UF2)
>>> microcontroller.reset()
```

jepler, 2023-08-06:
Good catch. Can you prepare a pull request? It looks like all ports should probably be reviewed but if your PR is just for one port that's fine.

bcr, 2023-08-06:
OK, my 124 checks seem to have passed.

bcr, 2023-08-06:
The following probably need review, I did not find any other places. I can make these changes if you like but I have no testing environment.

https://github.com/adafruit/circuitpython/blob/6b36bf59e30a4d3d93ec9df1f92ae8adc5fe2094/ports/cxd56/common-hal/microcontroller/__init__.c#L74
https://github.com/adafruit/circuitpython/blob/272a2dcdb0230cd13a6c7cc974805a919ef13a8d/ports/mimxrt10xx/common-hal/microcontroller/__init__.c#L70
https://github.com/adafruit/circuitpython/blob/272a2dcdb0230cd13a6c7cc974805a919ef13a8d/ports/stm/common-hal/microcontroller/__init__.c#L81

closed 2023-08-07 by jepler

bcr, 2024-01-17:
@jepler When will this ship? I think this was merged to main but not to the 8.2.x branch.

PR #8264 (2023-08-07): Add RUNMODE_UF2 handling for SAMD
Fixes #8263