adafruit/circuitpython #8268: Example for SSL

ams1, 2023-08-07:
Hi,

Thank you for this AWESOME initiative!

Any plan to have an example for SSL http server?

zap8600, 2023-08-07:
So we just need to modify the `espressif` port to check for the argument?

anecdata, 2023-08-08:
There's presumably some logic needed, predicated on `server_side=True`. But it is odd that the exception is `Invalid socket for TLS`, which I think indicates the socket isn't `SOCK_STREAM`, though the library sets it when it creates the socket to listen to.

dhalbert, 2023-08-11:
We could backport this to 8.2.x if it gets figured out soon.

michalpokusa, 2023-11-05:
Is there any progress on fixing/updating ESPs to allow HTTPS? I would like to make an update to `adafruit_httpserver` that allows HTTPS, but currently for all I know only RPi Pico W supports what is needed for it to work.

zap8600, 2023-11-29:
Why don't we just add
```c
    ret = mbedtls_ssl_config_defaults(&o->conf,
        server_side ? MBEDTLS_SSL_IS_SERVER : MBEDTLS_SSL_IS_CLIENT,
        MBEDTLS_SSL_TRANSPORT_STREAM,
        MBEDTLS_SSL_PRESET_DEFAULT);
    if (ret != 0) {
        goto cleanup;
    }
```
to [`common_hal_ssl_sslcontext_wrap_socket`](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/common-hal/ssl/SSLContext.c#L41)? Is there more that's supposed to be done? I'll make a fork and test this.

tannewt, 2024-02-14:
@michalpokusa or @anecdata did you try @zap8600's suggestion?

jepler, 2024-02-15:
I don't think calling mbedtls APIs here will work, the espressif SSLContext is using ep-idf calls like `esp_tls_init`.

It's tempted to think we could/should use mbedtls API calls (not esp-idf calls) for both espressif & raspberrypi rp2 (I think this is what micropython does!) but that would be a big change.

closed 2024-02-16 by dhalbert

anecdata, 2024-02-16:
Does anyone have working test code for this on ESP now? I'm getting `RuntimeError: Invalid socket for TLS` using code from `https://github.com/ide/circuitpython-https-server` updated for latest `adafruit_httpserver` (which now gets `MemoryError` on RPi. (I may be doing something wrong)

michalpokusa, 2024-02-16:
> Does anyone have working test code for HTTPS Server for `espressif` with this PR? I'm getting `RuntimeError: Invalid socket for TLS` using code from `https://github.com/ide/circuitpython-https-server` updated for latest `adafruit_httpserver`. (I may be doing something wrong)
> 
> i.e., how do we know this is completed?

I also started trying HTTPS with `adafruit_httpserver` on ESP32-S2 TFT Feather. I get exactly the same error, specifically it gets raised by `Server._sock.accept()` in `Server.poll()`.

anecdata, 2024-02-16:
client side (essentially same output for ESP32-S3 and Pico W:
```
% curl -v --insecure https://192.168.6.243
*   Trying 192.168.6.243:443...
* Connected to 192.168.6.243 (192.168.6.243) port 443
* ALPN: curl offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* Recv failure: Connection reset by peer
* LibreSSL/3.3.6: error:02FFF036:system library:func(4095):Connection reset by peer
* Closing connection
curl: (35) Recv failure: Connection reset by peer
```

anecdata, 2024-02-16:
@jepler works for me now on `espressif`, and it's plenty fast... Thanks!

michalpokusa, 2024-02-16:
@jepler I also tested on my MatrixPortal S3, works very fast, I also tested with Websockets and `wss://`.

But I was unable to make it work with ESP32-S2 Feather TFT, I can start the server but when trying to connect I get an empty `MemoryError:`. Seems strange considering that both have 2MB of RAM, although MatrixPortal has SRAM and Feather has PSRAM, would that make any difference?

anecdata, 2024-02-17:
I get the same on QT Py ESP32-S3 (N4R2), no dual http/https servers (just https):

server:
```
Traceback (most recent call last):
  File "adafruit_httpserver/server.py", line 404, in poll
MemoryError: 
Traceback (most recent call last):
  File "code.py", line 61, in <module>
  File "adafruit_httpserver/server.py", line 450, in poll
  File "adafruit_httpserver/server.py", line 404, in poll
MemoryError: 
```

client:
```
$ curl -iL -v --insecure https://192.168.6.180?HTTPS
*   Trying 192.168.6.180:443...
* Connected to 192.168.6.180 (192.168.6.180) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* Recv failure: Operation timed out
* LibreSSL/3.3.6: error:02FFF03C:system library:func(4095):Operation timed out
* Closing connection 0
curl: (35) Recv failure: Operation timed out
```

michalpokusa, 2024-02-17:
If RPicoW with 264kB can handle HTTPS, maybe there is something else here that needs to be fixed? It is definitely better but still not completely working.

Would it be worth re-opening this issue?

anecdata, 2024-02-17:
I think it needs an issue, either re-open or new. Some difference in S2 vs. S3, and also https://PicoW breaks with latest everything (unless you've gotten it to work?

michalpokusa, 2024-02-17:
I double checked and you are right, RPicoW now has the same problem as Feather, which is `MemoryError:`.

I assumed that considering that only the espressif port changed, code would still work on raspberry.

michalpokusa, 2024-02-18:
After updating to CP 9.0.0-beta.1 i get a different error:
[image]

This does not mean that the `MemoryError` is gone, as it might appear later, but something else changed that unables even ESP32-S3 boards like MatrixPortal S3 t ostart a HTTPS server.

anecdata, 2024-02-20:
Pico W HTTPS Server seems to have broken with the `MemoryError` somewhere between 8.0.0 and 8.1.0.

tannewt, 2024-02-20:
> iirc S2 has 320KB SRAM, and S3 has ~540KB~ 512KB SRAM, with various claims to that memory from esp-idf/rtos/etc, and maybe some things from CP core too. I don't fully understand the new CP 9 memory model.

CP will try to allocate to PSRAM first now in beta.1+. (Not exactly sure when I changed it.) The IDF can allocate to PSRAM now too by default but needs to request it explicitly. Without an explicit request, then internal memory will be used first.

PR #8926 (2024-02-16): mbedtls: move to shared-module
this enables the implementation to be shared among ports, and switches away from using espressif's tls wrapper layer.

code size seems about the same.

this probably fixes #8268 but I didn't test that.

I did test that client mode code worked (on pyportal esp32s3 using the script from #8910)