adafruit/circuitpython #8272: Problem with board pin assignments on Challenger RP2040 LTE

psturgeon54, 2023-08-09:
### CircuitPython version

```python
Version:  adafruit-circuitpython-challenger_rp2040_lte-en_GB-8.2.2.uf2
```

### Code/REPL

```python
import board
dir(board)
```

### Behavior

Adafruit CircuitPython 8.2.2 on 2023-08-01; Challenger RP2040 LTE with rp2040
>>> import board
>>> dir(board)
['__class__', '__name__', 'A0', 'A1', 'A2', 'A3', 'A4', 'A5', 'D0', 'D1', 'D10', 'D11', 'D12', 'D13', 'D5', 'D6', 'D9', 'GP0', 'GP1', 'GP10', 'GP11', 'GP12', 'GP15', 'GP16', 'GP17', 'GP19', 'GP2', 'GP21', 'GP22', 'GP23', 'GP24', 'GP25', 'GP26', 'GP27', 'GP28', 'GP29', 'GP3', 'GP4', 'GP5', 'GP6', 'GP7', 'GP8', 'GP9', 'I2C', 'LED', 'MISO', 'MOSI', 'RX', 'SARA_BTN', 'SARA_CTS', 'SARA_PWR', 'SARA_RST', 'SARA_RTS', 'SARA_RX', 'SARA_TX', 'SCK', 'SCL', 'SDA', 'SPI', 'TX', 'UART', 'board_id']
>>> 

### Description

I am trying to use the Challenger RP2040 LTE pin labelled SDI on the PCB, but I am having trouble.

According to the Challemger RP2040 LTE circuit, this connection goes to the RP2040 chip on its pin 31 which is GPIO20. Using the board module in the latest release of CircuitPython for this board, I have created a simple test program which assigns all the external board pins as inputs with pull-up.  The test program then displays on REPL the pin ID of any pin which is grounded. This works for every pin on the board, except the pin labelled SDI (i.e. the pin between the ones labelled SDO and RX)

I have a hunch that the board module content for this particular pin may be incorrect, possibly because board does not have an SDI or GP20 attribute

anecdata, 2023-08-09:
On this pin diagram, `SDI` is `IO24` / `GP24`. `IO20`/ `GP20`  isn't exposed. Are you looking at something different?
https://ilabs.se/challenger-rp2040-lte-datasheet/

psturgeon54, 2023-08-09:
Here is the circuit from iLabs for the Challenger RP2040 LTE:
https://gitlab.com/invectorlabs/hw/challenger-rp2040-lte/-/blob/main/V0.2/challenger-rp2040-lte.pdf?ref_type=heads

On the circuit, the connection labelled SDI on header JP2 goes to the RP2040 pin shown as GPIO20.
I see that this pin ihas no assignment in the microcontroller.pin mapping table.
microcontroller.pin.GPIO20 has no assignment

psturgeon54, 2023-08-09:
I suspect there may be other errors in the pinout diagram shown in https://ilabs.se/challenger-rp2040-lte-datasheet/
I tried raising this matter with Pontus Odberg of ILabs, but he hasn't yet replied to an email I sent him with a detailed explanation of why I think the labelling of one of the other pins is wrong.  I haven't had time to check through the details of every single pin, but I suppose that various issues will be apparent if you consult the circuit diagram for all the pins on JP1 and JP2 and refer to the RP2040 datasheet section 1.4 of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf

I can well understand how errors might creep in where there is such a profusion of pin names, pin IDs and different funcions that the RP2040 can map to each of its GPIO pins

anecdata, 2023-08-09:
Ah, I see. I hadn't found the schematic. On the schematic, pins 18 and 24 are unconnected.

You can use `microcontroller.pin.GPIO20` in CircuitPython in code a in place of `board.*`, to confirm that the schematic is correct and the pin diagram is wrong.

tannewt, 2023-08-09:
FYI @PontusO

anecdata, 2023-08-09:
The scheamtic is indeed wrong.
```py
>>> led = digitalio.DigitalInOut(microcontroller.pin.GPIO20)
>>> led.direction = digitalio.Direction.OUTPUT
>>> 
>>> led.value = True
>>> led.value = False
>>> 
>>> led = digitalio.DigitalInOut(microcontroller.pin.GPIO24)
>>> led.direction = digitalio.Direction.OUTPUT
>>> 
>>> led.value = True
>>> led.value = False
```
The second set of code, using pin 24, toggles an external LED connected to the `SDI` pin.

psturgeon54, 2023-08-09:
Thanks for the tips.  I had assumed that the circuit schematic was the gold-standard for what was actually connected to what.  Since the schematic showed the JP1 SDI pin connected to RP2040 GPIO20 and GP20 was not in the list of board.* assignments you gave, I figured that was why my simple test program couldn't get that pin to work as an input using board.*.    I will have a crack at what you suggest tomorrow.  Clearly something is amiss in either the schematic, the Challenger LTE pinout diagram or the board module.  We shall see...

psturgeon54, 2023-08-10:
Having updated my test program as you suggested, I can now confirm that the pin labelled SDI on the Challener RP2040 LTE is indeed microcontroller.pin.GPIO20.  I suppose this proves that the schematic is correct and that the pinout diagram has at least one error.
According to the schematic RP2040 GPIO24 is unconnected.  I presume then that the mapping should be something like

microcontroller.pin.GPIO20         board.SDI  board.GP20

anecdata, 2023-08-10:
Do you have a Mk I or a Mk II board? I have a Mk I.

psturgeon54, 2023-08-10:
We purchased ours form Pi Hut in January this year.  I can see from the order confirmation email that the order was for the MK II version.

However, I have looked very closely at the PCB under our workshop pcb microscope, and I can't see anything on the PCB silk-screen which specifically indentifies it as being a Mk II.

anecdata, 2023-08-10:
That could account for it, we probably need iLabs to clarify. Assuming the versions are different, that will complicate what to do about the board definition.

anecdata, 2023-08-10:
Reply on iLabs Discord indicates there was a change between board revisions
https://discord.com/channels/731576543050203147/902533648874618880/1139222674229432370

jepler, 2023-08-10:
That link doesn't appear to work if you're not already a member of the discord. If there's any additional useful content in the message can you add it here by copypaste?

anecdata, 2023-08-10:
> iLabs Electronics — Today at 10:44 AM
I saw I was tagged on github for this issue. I'm on vacation right now but will look into it on monday.
But as a quick response to this, yes we did make a revision to the board with regards to io24 and io20. The first version was incorrect making it difficult to use SPI (io24 can't be used as SDI).

psturgeon54, 2023-08-10:
I'm still very much of a beginner with CircuitPython, so your handy tip about how to use microcontroller.pin.GPIOx instead of board.* was a big help.  

Since you are clearly some rungs higher up than me on the Challenger LTE, CircuitPython and github ladders, I am wondering whether you might like to discuss a project I'm working on that uses Challenger LTE to run a Blynk client.  I presume that this isn't the place to have such a discussion.  If you are interested, my email is psturgeon@agiltechnik.cloud.

anecdata, 2023-08-10:
Adafruit Discord is a great place to discuss projects like that: https://adafru.it/discord
It's a community of mostly hobbyists and volunteers, very helpful on a variety of topics.

psturgeon54, 2023-08-11:
Thanks again.  I imagine there are quite a few hobbyists who know an awful lot more than me about the relevant issues.  I'll be along to discord in a wee while.

PR #9249 (2024-05-15): correction in pin assignment for Challenger RP2040 LTE
the board's [v0.2 revision](https://gitlab.com/invectorlabs/hw/challenger-rp2040-lte/-/blob/main/V0.2/challenger-rp2040-lte.pdf?ref_type=heads) moved SDI (MISO) from GP24 to GP20 for the SPI0 bus.

there shouldn't be any backwards compatibility issues since on the [rp2040 datasheet, table 2](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf), GP24 can not be used for SPI0 (only SPI1) so SPI on any of the old revisions of the board is non-functional since all the other SPI pins are on SPI0.

Closes: #8272
closed 2024-05-15 by dhalbert