adafruit/circuitpython #8275: CP 8.2.2: ESP32-S2 web workflow do not work!!

casainho, 2023-08-11:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.2 on 2023-07-31; S2Mini with ESP32S2-S2FN4R2
Board ID:lolin_s2_mini
UID:84722ED59B67
```

### Code/REPL

```python
import time

while True:
    print("Hello World!")
    time.sleep(0.5)
```

### Behavior

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Hello World!
Hello World!
Hello World!
Hello World!

### Description

I am trying to use the web workflow as explained on the documentation: https://docs.circuitpython.org/en/latest/docs/workflows.html#web

I used a basic code for testing. The ESP32-S2 is not connecting to my router and so the web workflow do not work. A few previous versions of CP, it was working well as expected.

[image]

[image]

[image]

### Additional information

I am being using the [Lolin S2 mini board](https://www.wemos.cc/en/latest/s2/s2_mini.html) and building [this displays](https://opensourceebike.github.io/build_EBike_Bafang_M500/build_display/build_display.html):

[image]

[image]

tannewt, 2023-08-11:
Are you able to get the CircuitPython serial output on the display? It'll show wifi status in the top bar. It looks like the serial terminal in vscode doesn't show it.

casainho, 2023-08-12:
I tested again on new boards, both with ESP32-S2 and ESP32-S3, with the latest CP 8.2.3. It works well as expected on both boards. Sorry, I now think I did some mistake, maybe a wrong password to my wifi router.

[image]

[image]

closed 2023-08-12 by casainho

casainho, 2023-08-19:
I got the issue again. It happens if I am using my home wifi but not if I used my 4G portable router.

But then I did remember I am using a wifi repeater - and I tested that if CP board is configured to connect to home main router it works, but do not work if I connect to the wifi repeater network. Maybe something changed on the wifi repeater, as I didn't had this issue previously...