adafruit/circuitpython #8278: SPI.deinit() crash

bill88t, 2023-08-12:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.3 on 2023-08-11; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
import board
a = board.SPI()
a.deinit()
```

### Behavior

[16:33:14.473] Disconnected
[16:33:15.474] Warning: Could not open tty device (No such file or directory)
[16:33:15.474] Waiting for tty device..
[16:33:17.477] Connected

Auto-reload is off.
Running in safe mode! Not running saved code.

ladyada, 2023-08-13:
well you def shouldnt be able to de'init it since its still being used by the TFT :) but should not hardcrash either

bill88t, 2023-08-13:
Perhaps it would be best to have a check inside of the SPI type that prohibits it from being deinitialised. Having that exposed only to the C api could allow circuitpy to alloc it for "system usage" and prohibit it's deinit. Shall I go and give it a try?

ladyada, 2023-08-13:
gooooood question - im going to let @tannewt or @jepler decide who/where it should be fixed :)

tannewt, 2023-08-14:
Seems to me that the best way to fix is to have displayio check for deinit SPI before using it. If it is, then displayio can deinit itself.

PR #9568 (2024-08-28): Avoid crashing when display components are deinitialized
Now, try_lock (SPI & I2C) & begin_transaction (display bus core) will check that the related objects are still valid first; if they are not, the lock/begin transaction will fail by returning false, rather than "other things" such as raising a Python exception where it is not permitted, accessing invalid memory, etc.

I tested by running a version of the #9426 reproducer on a Raspberry Pi Pico. It was able to complete & soft reset multiple times with no error. I also made sure that basic display functionality still worked on a Feather ESP [... cut]
commit 19e5cf3d (2024-08-28): Avoid crashing when display components are deinitialized

Now, try_lock (SPI & I2C) & begin_transaction (display bus core) will check
that the related objects are still valid first; if they are not,
the lock/begin transaction will fail by returning false, rather than
"other things" such as raising a Python exception where it is not
permitted, accessing invalid memory, etc.

Closes #8278 and Closes #9426
closed 2024-08-28 by tannewt