adafruit/circuitpython #8284: Adafruit Circuitpython trinkey 5022

PythonAmateur742, 2023-08-15:
### CircuitPython version

```python
adafruit-circuitpython-adafruit_proxlight_trinkey_m0-en_US-8.2.3.uf2 with proximity trinkey 5022
```

### Code/REPL

```python
# Copied and pasted the code form here 
# https://learn.adafruit.com/adafruit-proximity-trinkey/proximity-spacebar-game

# and here:
# https://learn.adafruit.com/adafruit-proximity-trinkey/proximity-midi-controller

# The last one to worked for a bit, but not all the time. It didn't run in Thonny though. 
# I had to rename it to code.py, include all the libraries in /lib and it worked.
# When I included the adafruit hid library, it stopped working.
```

### Behavior

In Thonny:
Unable to connect to COM15: port not found

Or similar issues. Like COM15 busy/occupied/no permission

When there was an error introduced into the code (sounds on purpose, but it was accidental), the error wasn't reported by thonny.
Error on line xx, blabla was expected, but it didn't do that.

I reverted to version 7 and Thonny and the trinkey behaved normally.

DavePutz, 2023-08-26:
I was able to reproduce this on a Metro M0 Express. But it may be a Thonny issue, as there is a stack trace in the Thonny backend log. Using a terminal emulator such as TeraTerm to connect to the board shows the correct responses. I have opened an issue with Thonny as well (https://github.com/thonny/thonny/issues/2914).

dhalbert, 2023-11-07:
Another Thonny/SAMD21 bug: https://github.com/adafruit/circuitpython/issues/8038