adafruit/circuitpython #8294: MatrixPortal S3 Hard Fault to Safemode 8.2.3

DJDevon3, 2023-08-18:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.3 on 2023-08-11; Adafruit MatrixPortal S3 with ESP32S3
Board ID:adafruit_matrixportal_s3
Library Bundle: adafruit-circuitpython-bundle-8.x-mpy-20230815
```

### Code/REPL

```python
import displayio
import board
import rgbmatrix
import framebufferio

bit_depth = 1
base_width = 64
base_height = 32
chain_across = 2
tile_down = 0
serpentine = True

width = base_width * chain_across
height = base_height * tile_down

addr_pins = [board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD]
rgb_pins = [
    board.MTX_R1,
    board.MTX_G1,
    board.MTX_B1,
    board.MTX_R2,
    board.MTX_G2,
    board.MTX_B2,
]
clock_pin = board.MTX_CLK
latch_pin = board.MTX_LAT
oe_pin = board.MTX_OE

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
                width=width,
                height=height,
                bit_depth=bit_depth,
                rgb_pins=rgb_pins,
                addr_pins=addr_pins,
                clock_pin=clock_pin,
                latch_pin=latch_pin,
                output_enable_pin=oe_pin,
                tile=tile_down, serpentine=serpentine,
            )
display = framebufferio.FramebufferDisplay(matrix, auto_refresh=False)

# Create two lines of text to scroll. Besides changing the text, you can also
# customize the color and font (using Adafruit_CircuitPython_Bitmap_Font).
# To keep this demo simple, we just used the built-in font.
# The Y coordinates of the two lines were chosen so that they looked good
# but if you change the font you might find that other values work better.
line1 = adafruit_display_text.label.Label(
    terminalio.FONT,
    color=0xff0000,
    text="This scroller is brought to you by CircuitPython RGBMatrix")
line1.x = display.width
line1.y = 8

line2 = adafruit_display_text.label.Label(
    terminalio.FONT,
    color=0x0080ff,
    text="Hello to all CircuitPython contributors worldwide <3")
line2.x = display.width
line2.y = 24

# Put each line of text into a Group, then show that group.
g = displayio.Group()
g.append(line1)
g.append(line2)
display.show(g)

# This function will scoot one label a pixel to the left and send it back to
# the far right if it's gone all the way off screen. This goes in a function
# because we'll do exactly the same thing with line1 and line2 below.
def scroll(line):
    line.x = line.x - 1
    line_width = line.bounding_box[2]
    if line.x < -line_width:
        line.x = display.width

# This function scrolls lines backwards.  Try switching which function is
# called for line2 below!
def reverse_scroll(line):
    line.x = line.x + 1
    line_width = line.bounding_box[2]
    if line.x >= display.width:
        line.x = -line_width

# You can add more effects in this loop. For instance, maybe you want to set the
# color of each label to a different value.
while True:
    scroll(line1)
    scroll(line2)
    #reverse_scroll(line2)
    display.refresh(minimum_frames_per_second=0)
```

### Behavior

```py
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Fault detected by hardware.
Please file an issue with your program at https://github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

### Description

Initially got the script up and running with little effort. Then suddenly it began to hard fault so many times that Windows refused to recognize if I plugged the MatrixPortal S3 by itself not connected to any panels.  Would not show up as CIRCUITPY.  Restarted windows and it recognized again however it hard faults every single time now no matter what I do.

Possible user error as it's my first time using the MatrixPortal S3.  Power delivery is not an issue with a 5V 20A meanwell PSU only connected to 2 matrix panels. Again, hard fault occurs regardless if it's connect to the panels or not.

### Additional information

I was able to get it to display a bmp with imageload, tilegrid, and labels like a normal displayio project but things went downhill quickly after it began hard faulting.

Used a combination/mashup of learn guide code from:
https://learn.adafruit.com/rgb-led-matrices-matrix-panels-with-circuitpython/advanced-multiple-panels
and
https://learn.adafruit.com/rgb-led-matrices-matrix-panels-with-circuitpython/example-simple-two-line-text-scroller

makermelissa, 2023-08-18:
I noticed hard faulting as well while working on my message board project, but not as bad as described above. I thought I'd add some details from my experience in case that helps. It happens after about every few resets and really has been an issue since the board definition was added. If code is left running for me, there are no issues.

DJDevon3, 2023-08-18:
@makermelissa Yes same, during the process it would hard fault occasionally and reset would help. At some point hard reset happened more and more frequently until it hard faults every single time and cannot be used... at least not with matrix panel related code.  I have not tried to go back and do a basic hello world or non-matrix panel related code.  

I think Windows itself was detecting the continual hard resets and at some point refused to allow CIRCUITPY to even connect. No Com port in device manager and no CIRCUITPY driver regardless of power cycles or hard resets.  A reboot of the entire PC did work, so might be Windows related to some extent if the issue continually happens.  I'd say around the 50th hard reset within a couple hours it refused to detect.  Had to run device cleanup and restart.  

I have no other CIRCUITPY devices connected to the PC except the MatrixPortal S3.

Was running into hard faults a ton so I just disconnected the board.  It's kind of unusable in its current state.  There's a hard fault bug somewhere. I have absolutely no clue or advice where it's coming from.

DJDevon3, 2023-08-18:
Just realized it might help if I specify I'm using it with 4x 64x32 5mm pitch panels with a 5v 18A meanwell PSU.

DJDevon3, 2023-08-23:
Since finding the adafruit_requests bug I reverted the MatrixPortal S3 down to 8.2.0. I realize there are some bug fixes for it since then.  Kind of stuck on this one for a little bit for multiple reasons.

Was able to get at least 2 panels working. Could not get 3 or 4 working using any setup of across, down, or serpentine.  Possibly due to bad cable as panels 3 & 4 were artifacting heavily.  Attempted to DIY a longer cable with 5 feet of ribbon cable from digikey and ended up breaking the IDC connector instead. I have now purchased longer cables with IDC connectors attached.  Never try to pry open and reuse IDC plugs, it won't end well, live and learn.

Part of the reason this code fails is due to variable names being the same as parameter names. I'm pretty sure I ripped most of this from learn guide code. That might be a problem and a learn guide might need updating.
```py
width = base_width * chain_across
height = base_height * tile_down

matrix = rgbmatrix.RGBMatrix(
                width=width,
                height=height,
                bit_depth=bit_depth,
                rgb_pins=rgb_pins,
                addr_pins=addr_pins,
                clock_pin=clock_pin,
                latch_pin=latch_pin,
                output_enable_pin=oe_pin,
                tile=tile_down, serpentine=serpentine,
            )
```
This fails because matrix parameters are always equaling themselves and not the variables set above them due to variable name conflicts. Inputting width=64, height=32, etc.. manually works. Prepending all variable names (shown below) with unique names like panel_variable also works.

```py
panel_width = panel_base_width * panel_chain_across
panel_height = panel_base_height * panel_tile_down

matrix = rgbmatrix.RGBMatrix(
                width=panel_width,
                height=panel_height,
                bit_depth=panel_bit_depth,
                rgb_pins=panel_rgb_pins,
                addr_pins=panel_addr_pins,
                clock_pin=panel_clock_pin,
                latch_pin=panel_latch_pin,
                output_enable_pin=panel_oe_pin,
                tile=panel_tile_down, 
                serpentine=panel_serpentine,
            )
```
Figured out partly why it was crashing so much. While a parameter name like width= normally shouldn't be a problem... it absolutely was, as if it was being recognized as a variable instead of a parameter.  So width=width was just equaling itself, not the variable set above, and failing. I have no idea how/why this might happen other than not providing the variable with a unique name that doesn't = the parameter name.

No progress on tracking down the hard fault.

dhalbert, 2023-08-23:
Tagging @PaintYourDragon, since this might be Protomatter-related.

DJDevon3, 2023-08-23:
I saw the update by @makermelissa for swapping the ADDRA and ADDRD pins.  I swapped them manually in software last night and that didn't help.  There are also a couple other updates that might help fix this bug in 8.2.3 or 8.2.4 

Since I'm still having hard faults, by reverting to 8.2.0 to cut the adafruit_requests bug out is of no help since 8.2.0 is worse than 8.2.3 for the matrixportal s3 running matrix panel code.  Can't even get all the panels up and working so reverting CP was a pointless endeavor... other than absolutely confirming the adafruit_requests bug 100% does affect this board as well.

Will update to 8.2.4 today and go from there.  In terms of the matrixportal s3 finding the hard fault when used with matrix panel libraries is far more important.

DJDevon3, 2023-08-23:
Started modifying the code a little in an attempt to narrow down the bug.  Still hard faults with this new code.  Attempting to bisect a little today.

```py
Adafruit CircuitPython 8.2.4 on 2023-08-22; Adafruit MatrixPortal S3 with ESP32S3
Board ID:adafruit_matrixportal_s3
```

```py
import board
import displayio
import rgbmatrix
import framebufferio
import terminalio
from adafruit_display_text import label

panel_bit_depth = 1
panel_base_width = 64
panel_base_height = 32
panel_chain_across = 2
panel_tile_down = 0
panel_serpentine = False

panel_width = panel_base_width * panel_chain_across
panel_height = panel_base_height * panel_tile_down

panel_addr_pins = [board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD]
panel_rgb_pins = [
    board.MTX_R1,
    board.MTX_G1,
    board.MTX_B1,
    board.MTX_R2,
    board.MTX_G2,
    board.MTX_B2,
]
panel_clock_pin = board.MTX_CLK
panel_latch_pin = board.MTX_LAT
panel_oe_pin = board.MTX_OE

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
                width=panel_width,
                height=panel_height,
                bit_depth=panel_bit_depth,
                rgb_pins=panel_rgb_pins,
                addr_pins=panel_addr_pins,
                clock_pin=panel_clock_pin,
                latch_pin=panel_latch_pin,
                output_enable_pin=panel_oe_pin,
                tile=panel_tile_down,
                serpentine=panel_serpentine,
            )
display = framebufferio.FramebufferDisplay(matrix, rotation=0)
print(f"Panel Width: {panel_width}")
print(f"Panel Height: {panel_height}")

# Quick Colors for Labels
text_black = 0x000000
text_blue = 0x0000FF
text_cyan = 0x00FFFF
text_gray = 0x8B8B8B
text_green = 0x00FF00
text_lightblue = 0x90C7FF
text_magenta = 0xFF00FF
text_orange = 0xFFA500
text_purple = 0x800080
text_red = 0xFF0000
text_white = 0xFFFFFF
text_yellow = 0xFFFF00

hello_label = label.Label(terminalio.FONT)
hello_label.anchor_point = (0.5, 1.0)
hello_label.anchored_position = (panel_width/2, 15)
hello_label.scale = (1)
hello_label.color = text_white

# Load Bitmap to tile grid first (Background layer)
DiskBMP = displayio.OnDiskBitmap("/images/Adafruit_32px_RGB_Matrix_Logo.bmp")
tile_grid = displayio.TileGrid(
    DiskBMP,
    pixel_shader=DiskBMP.pixel_shader,
    width=1,
    height=1,
    tile_width=panel_width,
    tile_height=panel_height)

# Create subgroups
text_group = displayio.Group()
text_group.append(tile_grid)
main_group = displayio.Group()

# Add subgroups to main display group
main_group.append(text_group)

# Label Display Group (foreground layer)
text_group.append(hello_label)
display.show(main_group)

hello_label.text = "MatrixPortal S3"
```

```py
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

DJDevon3, 2023-08-23:
It will hard fault if tile does not =1.  Try =0 or =2 and it hard faults. Regardless if I use 64x32 or 128x32

[Adafruit_32px_RGB_Matrix_Logo.zip](https://github.com/adafruit/circuitpython/files/12423068/Adafruit_32px_RGB_Matrix_Logo.zip)

```py
# SPDX-FileCopyrightText: 2022 DJDevon3 for Adafruit Industries
#
# SPDX-License-Identifier: MIT
# Used with Adafruit Feather ESP32-S3
# Coded for Circuit Python & 8.0.0-beta.1 libraries

import time
from math import sin
import board
import displayio
import rgbmatrix
import framebufferio
import adafruit_imageload
import terminalio
from adafruit_display_text.label import Label
 
displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=64, height=32, bit_depth=1,
    rgb_pins=[board.MTX_R1, board.MTX_G1, board.MTX_B1, board.MTX_R2, board.MTX_G2, board.MTX_B2],
    addr_pins=[board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD],
    clock_pin=board.MTX_CLK, latch_pin=board.MTX_LAT, output_enable_pin=board.MTX_OE, tile=1, serpentine=True,
    doublebuffer=True)

# Associate the RGB matrix with a Display so we can use displayio
display = framebufferio.FramebufferDisplay(matrix, auto_refresh=False)

# Quick Colors for Labels
text_black = 0x000000
text_blue = 0x0000FF
text_cyan = 0x00FFFF
text_gray = 0x8B8B8B
text_green = 0x00FF00
text_lightblue = 0x90C7FF
text_magenta = 0xFF00FF
text_orange = 0xFFA500
text_purple = 0x800080
text_red = 0xFF0000
text_white = 0xFFFFFF
text_yellow = 0xFFFF00

g = displayio.Group()
b, p = adafruit_imageload.load("images/Adafruit_32px_RGB_Matrix_Logo.bmp")
t = displayio.TileGrid(b, pixel_shader=p)
t.x = 20
g.append(t)

l = Label(text="MatrixPortal S3", font=terminalio.FONT, color=text_magenta, line_spacing=1.0)
g.append(l)

display.show(g)

target_fps = 10
ft = 1/target_fps
now = t0 = time.monotonic_ns()
deadline = t0 + ft

p = 1
q = 17
while True:
    tm = (now - t0) * 1e-9
    x = l.x - 1
    if x < -100:
        x = 63
    y =  round(12 + sin(tm / p) * 6)
    l.x = x
    l.y = y
    display.refresh(target_frames_per_second=target_fps, minimum_frames_per_second=0)
    while True:
        now = time.monotonic_ns()
        if now > deadline:
            break
        time.sleep((deadline - now) * 1e-9)
    deadline += ft
```

This works
```py
matrix = rgbmatrix.RGBMatrix(
    width=64, height=32, bit_depth=1,
    rgb_pins=[board.MTX_R1, board.MTX_G1, board.MTX_B1, board.MTX_R2, board.MTX_G2, board.MTX_B2],
    addr_pins=[board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD],
    clock_pin=board.MTX_CLK, latch_pin=board.MTX_LAT, output_enable_pin=board.MTX_OE, tile=1, serpentine=True,
    doublebuffer=True)
```
[image]

This hard faults
```py
matrix = rgbmatrix.RGBMatrix(
    width=64, height=32, bit_depth=1,
    rgb_pins=[board.MTX_R1, board.MTX_G1, board.MTX_B1, board.MTX_R2, board.MTX_G2, board.MTX_B2],
    addr_pins=[board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD],
    clock_pin=board.MTX_CLK, latch_pin=board.MTX_LAT, output_enable_pin=board.MTX_OE, tile=0, serpentine=True,
    doublebuffer=True)
```
The only difference is tile=0

Also it appears as if every panel is being duplicated then shifted out of frame.  If I choose 64x32 I get 2 mirrored panels Instead of 1 blank panel and 1 64x32.

DJDevon3, 2023-08-25:
Updated to 
```py
Adafruit CircuitPython 9.0.0-alpha.1-8-gaf91625fb-dirty on 2023-08-23; Adafruit MatrixPortal S3 with ESP32S3
Board ID:adafruit_matrixportal_s3
```
Working code:
```py
# SPDX-FileCopyrightText: 2023 DJDevon3
#
# SPDX-License-Identifier: MIT
# Used with Adafruit MatrixPortal S3
# Coded for Circuit Python 9.0 alpha

import time
from math import sin
import board
import displayio
import rgbmatrix
import framebufferio
import adafruit_imageload
import terminalio
from adafruit_display_text.label import Label
 
displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=128, height=64, bit_depth=2,
    rgb_pins=[board.MTX_R1, board.MTX_G1, board.MTX_B1, board.MTX_R2, board.MTX_G2, board.MTX_B2],
    addr_pins=[board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD],
    clock_pin=board.MTX_CLK, latch_pin=board.MTX_LAT, output_enable_pin=board.MTX_OE, tile=2, serpentine=True,
    doublebuffer=True)

# Associate the RGB matrix with a Display so we can use displayio
display = framebufferio.FramebufferDisplay(matrix, auto_refresh=False)

# Quick Colors for Labels
text_black = 0x000000
text_blue = 0x0000FF
text_cyan = 0x00FFFF
text_gray = 0x8B8B8B
text_green = 0x00FF00
text_lightblue = 0x90C7FF
text_magenta = 0xFF00FF
text_orange = 0xFFA500
text_purple = 0x800080
text_red = 0xFF0000
text_white = 0xFFFFFF
text_yellow = 0xFFFF00

g = displayio.Group()
b, p = adafruit_imageload.load("images/Adafruit_32px_RGB_Matrix_Logo.bmp")
t = displayio.TileGrid(b, pixel_shader=p)
t.x = 48
t.y = 15
g.append(t)

l = Label(text="MatrixPortal S3", font=terminalio.FONT, color=text_magenta, line_spacing=1.0)
g.append(l)

display.show(g)

target_fps = 10
ft = 1/target_fps
now = t0 = time.monotonic_ns()
deadline = t0 + ft

p = 1
q = 17
while True:
    tm = (now - t0) * 1e-9
    x = l.x - 1
    if x < -100:
        x = 127
    y =  round(12 + sin(tm / p) * 6)
    l.x = x
    l.y = y +20
    display.refresh(target_frames_per_second=target_fps, minimum_frames_per_second=0)
    while True:
        now = time.monotonic_ns()
        if now > deadline:
            break
        time.sleep((deadline - now) * 1e-9)
    deadline += ft
```

https://github.com/adafruit/circuitpython/assets/49322231/c494dcda-dc14-463b-ae45-a0d3d81b8baf

While I was able to get it to work it still exhibits the same hard faults if I deviate from these settings in almost any way.

makermelissa, 2023-09-01:
What happens f you leave the height out, but use tile set to 2? Apparently it's an optional parameter.

DJDevon3, 2023-09-02:
@makermelissa on the 9.0 alpha 1.8 build or 8.2.4?  I've since switched to the alpha build which works much better.  I can go back to 8.2.4 for more testing if needed. 

I still get intermittent hard faults on 9.0 alpha but it only when saving code.py.  A hard reset usually fixes it and will run indefinitely.  I do not get hard faults while the program is running, only happens on file save (using Mu). A good example of the hard fault during save can be seen in the live demo on this weeks [8/30/2023 Show & Tell](https://www.youtube.com/live/uKouVzFcVxM?feature=shared&t=976).  The full code I was running during that demo is now [available on my github](https://github.com/DJDevon3/My_Circuit_Python_Projects/tree/main/Boards/espressif/Adafruit%20MatrixPortal%20S3/128x64%20RGB%20Matrix). Used the latest 8.x PY libraries as of 8/30/2023 with that 9.0.0 alpha_1.8 build.

tannewt, 2023-09-27:
I think I've been chasing this bug. The short reasoning is that RGBMatrix uses the supervisor to allocate some memory. When it does, it doesn't hold onto the outer allocation object. Instead, when freeing a pointer, it looks up the allocation object again. This only works if the memory isn't moved by the supervisor (which it does.) This leaks supervisor allocations and the crash occurs when out of allocation objects.

I'm working on a fix now.

tannewt, 2023-09-29:
Hopefully fixed by #8446

DJDevon3, 2023-10-20:
@tannewt seems fixed in 8.2.7 ran it through some paces after the new release last night and no more hard faults. I cannot attest to issue 8444 though.

tannewt, 2023-11-01:
I think this was fixed by #8446

closed 2023-11-01 by tannewt