adafruit/circuitpython #8299: MatrixPortal S3 Fails with Learn Guide simpletest

DJDevon3, 2023-08-19:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.3 on 2023-08-11; Adafruit MatrixPortal S3 with ESP32S3
Board ID:adafruit_matrixportal_s3
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2020 Brent Rubell for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import os
import ipaddress
import ssl
import wifi
import socketpool
import adafruit_requests

# URLs to fetch from
TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"
JSON_QUOTES_URL = "https://www.adafruit.com/api/quotes.php"
JSON_STARS_URL = "https://api.github.com/repos/adafruit/circuitpython"

print("ESP32-S2 WebClient Test")

print(f"My MAC address: {[hex(i) for i in wifi.radio.mac_address]}")

print("Available WiFi networks:")
for network in wifi.radio.start_scanning_networks():
    print("\t%s\t\tRSSI: %d\tChannel: %d" % (str(network.ssid, "utf-8"),
                                             network.rssi, network.channel))
wifi.radio.stop_scanning_networks()

print(f"Connecting to {os.getenv('WIFI_SSID')}")
wifi.radio.connect(os.getenv("WIFI_SSID"), os.getenv("WIFI_PASSWORD"))
print(f"Connected to {os.getenv('WIFI_SSID')}")
print(f"My IP address: {wifi.radio.ipv4_address}")

ping_ip = ipaddress.IPv4Address("8.8.8.8")
ping = wifi.radio.ping(ip=ping_ip) * 1000
if ping is not None:
    print(f"Ping google.com: {ping} ms")
else:
    ping = wifi.radio.ping(ip=ping_ip)
    print(f"Ping google.com: {ping} ms")

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

print(f"Fetching text from {TEXT_URL}")
response = requests.get(TEXT_URL)
print("-" * 40)
print(response.text)
print("-" * 40)

print(f"Fetching json from {JSON_QUOTES_URL}")
response = requests.get(JSON_QUOTES_URL)
print("-" * 40)
print(response.json())
print("-" * 40)

print()

print(f"Fetching and parsing json from {JSON_STARS_URL}")
response = requests.get(JSON_STARS_URL)
print("-" * 40)
print(f"CircuitPython GitHub Stars: {response.json()['stargazers_count']}")
print("-" * 40)

print("Done")
```

### Behavior

```py
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESP32-S2 WebClient Test
My MAC address: ['obfuscated']
Available WiFi networks:
	SlurpNet		RSSI: -57	Channel: 10
Connecting to SlurpNet
Connected to SlurpNet
My IP address: obfuscated
Ping google.com: 0.0 ms
Fetching text from http://wifitest.adafruit.com/testwifi/index.html
----------------------------------------
This is a test of Adafruit WiFi!
If you can read this, its working :)
----------------------------------------
Fetching json from https://www.adafruit.com/api/quotes.php
----------------------------------------
[{'text': 'I don’t care that they stole my idea — I care that they don’t have any of their own', 'author': 'Nikola Tesla'}]
----------------------------------------

Fetching and parsing json from https://api.github.com/repos/adafruit/circuitpython
Traceback (most recent call last):
  File "adafruit_requests.py", line 515, in _get_socket
OSError: Failed SSL handshake

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 58, in <module>
  File "adafruit_requests.py", line 711, in get
  File "adafruit_requests.py", line 650, in request
  File "adafruit_requests.py", line 496, in _get_socket
RuntimeError: Sending request failed

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

Using Internet Test code from [MatrixPortal S3 learn guide](https://learn.adafruit.com/adafruit-matrixportal-s3/circuitpython-internet-test).
- succeeds SSL handshake with Adafruit HTTPS quotes url
- fails SSL handshake with HTTPS Github API

I have no problem viewing https://api.github.com/repos/adafruit/circuitpython with my browser. Shows all the JSON data fine. For whatever reason Github is failing to handshake with the MatrixPortal S3. 

### Additional information

This is only to report that a portion of code in a learn guide does not function correctly. I'm honestly not sure why it's failing.

Using adafruit_requests.mpy from adafruit-circuitpython-bundle-8.x-mpy-20230815

Matrix Portal is not connected to any matrix panels at this time. Using the board as I would a feather S3 with wifi.  

I can get it to ignore the error with try/except but that's not the point in the case of learn guide code.

DJDevon3, 2023-08-19:
Removed adafruit gets and focused only on the Github connection.  It's not connecting to that API url. Rest of the script seems to work fine. It's specifically the Github handshake attempt.

DJDevon3, 2023-08-19:
Here's a slimmer version that will fail gracefully during testing.

```py
import os
import time
import ipaddress
import ssl
import wifi
import socketpool
import adafruit_requests

pool = socketpool.SocketPool(wifi.radio)

JSON_STARS_URL = "https://api.github.com/repos/adafruit/circuitpython"

# Connect to Wi-Fi
print("\n====== MatrixPortal S3 WebClient Test ======")
print("Connecting to WiFi...")
requests = adafruit_requests.Session(pool, ssl.create_default_context())
while not wifi.radio.ipv4_address:
    try:
        wifi.radio.connect(os.getenv('WIFI_SSID'), os.getenv('WIFI_PASSWORD'))
    except ConnectionError as e:
        print("Connection Error:", e)
        print("Retrying in 10 seconds")
    time.sleep(10)
print("Connected!\n")

while True:
    try:
        response = requests.get(JSON_STARS_URL).json()
        stargazers_count = response['stargazers_count']
        print(f"CircuitPython GitHub Stars: {stargazers_count}")
    except (ValueError, RuntimeError, OSError) as e:
        print("Failed to get data, retrying\n", e)
        time.sleep(60)
        continue
    response = None

    print("Done")
```
With the OSError handler in the mix it spits out

```py
====== MatrixPortal WebClient Test ======
Connecting to WiFi...
Connected!

Failed to get data, retrying
 Sending request failed
Failed to get data, retrying
 Sending request failed
Failed to get data, retrying
 Sending request failed
Failed to get data, retrying
 Sending request failed
```
The script will attempt to connect indefinitely so no sense waiting for it to connect. It simply will not connect.  I have no idea why it isn't working with Github.  Have not attempted other API's.

DJDevon3, 2023-08-19:
Tried on an Adafruit ESP32-S2 Feather on 8.2.3 with the same result.

dhalbert, 2023-08-22:
Here is a roots.pem that I generated a couple of weeks ago using the procedure described in https://github.com/arduino/nina-fw/blob/master/README.md#build-a-new-certificate-list-based-on-the-google-android-root-ca-list.

[roots.pem.zip](https://github.com/adafruit/circuitpython/files/12413182/roots.pem.zip)

When I substitute this one for the one in `lib/certificates/data/roots.pem`, talking to github works again.
But I was unable to come up with an Adafruit IO example that breaks on the old roots.pem but not on this one, so I haven't made a PR yet.

I think we should stop depending on nina-fw as a submodule, and just keep our own roots.pem, documenting how we built it. We could even write makefile steps to do that and verify it.

RetiredWizard, 2023-08-23:
I've been building the Pico W with the latest main branch bits and this evening was getting the "Sending request failed" message with one of my test programs. After building with the roots.pem file Dan posted the test program is working again.

That being said, my test program was running fine when I built from main last night so I'm thinking the error only occurs when you hit a particular server somewhere in your path.

I'll keep using the new roots.pem and if I hit any failures with it I'll update here.

jepler, 2023-08-23:
The change to `nina-fw` that we think introduced this problem was initially done in the *8.2.x* branch. However, it probably started affecting *main* today when we merged #8317.

After nina-fw is updated, the fix in CircuitPython will probably initially be made in *8.2.x* and merged into *main* subsequent to that, though it depends on how Dan and/or Scott want to handle it.

RetiredWizard, 2023-08-23:
@DJDevon3 Just curious, did you try building the MatrixPortal S3 using the roots.pem file that Dan posted above?

DJDevon3, 2023-08-23:
No, building circuit python isn't something I've done in a long time. :/  I'm not good with make builds.

RetiredWizard, 2023-08-23:
This is using the latest bits from main so it has the 1.19.1 Micropython merge included.

[firmware.zip](https://github.com/adafruit/circuitpython/files/12423328/firmware.zip)

DJDevon3, 2023-08-23:
Is there a 9.0 alpha bundle?  I'm getting 
```py
Traceback (most recent call last):
  File "code.py", line 7, in <module>
ValueError: incompatible .mpy file
``` with 9.0.0-alpha.1-8-gaf91625fb-dirty

DJDevon3, 2023-08-23:
@dhalbert @RetiredWizard Unsure if this can be closed now or not. Up to you.

tannewt, 2023-08-24:
> Is there a 9.0 alpha bundle? I'm getting

Use the .py files for now. We don't want to may a mpy bundle until 1.20 is merged because it changes the mpy format again.

DJDevon3, 2023-08-25:
Thank you for the clarification. Already started transitioning the libraries on my 9.0 alpha boards to all py libraries and thus far everything is working as expected including the MatrixPortal S3.

bill88t, 2023-08-28:
Just noting, this is still an issue in the the 9.0 pre-alpha builds (9.0.0-alpha.1-13-gc395e1f9f).
I was reworking my wget and tried `wget https://raw.githubusercontent.com/adafruit/circuitpython/029c912bf0f19062426922962deaf12d9785c5bc/supervisor/shared/filesystem.c` to test how it coped with the filename.

```
Traceback (most recent call last):
  File "adafruit_requests", line 1, in _get_socket
OSError: Failed SSL handshake

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "ljinux", line 1, in fpexec
  File "<string>", line 14, in <module>
  File "driver_wifi", line 1, in get
  File "adafruit_requests", line 1, in get
  File "adafruit_requests", line 1, in request
  File "adafruit_requests", line 1, in _get_socket
RuntimeError: Sending request failed
```

Adafruit_requests is an mpy for this specific commit. I build the mpy's directly for the version of circuitpython loaded on the board.

RetiredWizard, 2023-08-28:
Yep, until the actual fix is decided on, if you want to use WiFi with the latest main, you need to manually replace lib/certificates/data/roots.pem with Dan's roots.pem file.

anecdata, 2023-08-28:
...or use:
```py
context = ssl.create_default_context()
context.load_verify_locations()
```
with a CA string or .pem file

anecdata, 2023-09-04:
> load_verify_locations for more than a single certificate?

I'm not sure if it's intended for a bundle (docs could be modified to indicate), just that it's sometimes used to get around a missing cert for a site.

Myoldmopar, 2023-09-04:
Thanks for everything you all are doing.  I struggled with this setting up a new Pico W today, trying to talk to the GitHub API.  Based on the comments above, I simply downgraded to CircuitPython 8.2.0 and it immediately started working.

dhalbert, 2023-09-04:
I plan to release an 8.2.x with updated root certificates soon.

DJDevon3, 2023-09-08:
Tested and confirmed fixed with 8.2.5 released today on the following boards
- Feather ESP32-S2 with 4mb psram
- Feather ESP32-S3 with 4mb psram
- MatrixPortal S3

Thank you @dhalbert, issue closed.

closed 2023-09-08 by DJDevon3

dhalbert, 2023-09-08:
@tyeth Could you open a new issue about the bundle loading? Also try it on regular CPython, and try just one cert and more than one.

tyeth, 2023-09-08:
Done, opened #8381