adafruit/circuitpython #8302: can't load adafruit_dotstar library on unexpectedmaker feather S2

gmparis, 2023-08-20:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.3 on 2023-08-11; FeatherS2 with ESP32S2
Board ID:unexpectedmaker_feathers2
```

### Code/REPL

```python
import adafruit_dotstar
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_dotstar.py", line 44, in <module>
ImportError: no module named 'adafruit_pypixelbuf'

### Description

Complains about missing adafruit_pypixelbuf. Tried replacing adafruit_dotstar.mpy with the .py version, as the code does not have the "py" in pixelbuf, but that did not help. Module must be part of what has been flashed to the device in 8.2.3.

### Additional information

Can't figure a workaround. Board will not work with dotstars.

DJDevon3, 2023-08-21:
The [check for adafruit_pypixelbuf was removed in March of 2022](https://github.com/adafruit/Adafruit_CircuitPython_DotStar/commit/fdc63e80404e7405489ac5922addf06a92093aba). You're using an outdated library bundle. Please grab a more recent [Bundle for Version 8.x](https://circuitpython.org/libraries)

gmparis, 2023-08-21:
I am using the one dated 20230815.

DJDevon3, 2023-08-21:
Then it should be using the [latest version of adafruit_dotstar](https://github.com/adafruit/Adafruit_CircuitPython_DotStar/blob/67f8c35e2ce4a8abc3cfb24ab5029b40a64857b8/adafruit_dotstar.py)... I would hope.

gmparis, 2023-08-21:
In the meantime I flashed the prior version of CircuitPy (8.2.2) and the
problem was rectified. Same library bundle.

DJDevon3, 2023-08-21:
If you perhaps had an adafruit_dotstar.py in the same /lib folder it would override the adafruit_dotstar.mpy from the bundle.

gmparis, 2023-08-21:
This was an otherwise new S2 I had in a drawer for a long time. I flashed
to 8.2.3 and it no longer ran the sample code that flashes the dotstar. At
this point /lib was empty, so I loaded the entire bundle into /lib. Still
broken at that point. You now have the entire history.

DJDevon3, 2023-08-21:
Only load libraries you currently need or actively use into /lib.  The fact that the UM Feather can store the entire bundle is a marvel in itself. That's not what should be done for the bundle... normally. ;) It's currently only one of very few boards with enough flash storage to do that.

gmparis, 2023-08-21:
Yes, that's normally what I do. I was being lazy this time. Are you saying
this is what caused the issue? I did the same thing for 8.2.2 and there is
no issue with it.

DJDevon3, 2023-08-21:
Most likely yes the .py file sounds like it was an old version that was overriding the newer .mpy from the bundle.

DJDevon3, 2023-08-21:
Figured out you did nothing wrong.  Unexpected Maker ships the board pre-installed with Circuit Python and an adafruit_dotstar.py in the root directory which will override any adafruit_dotstar.py or .mpy in the /lib folder.  That's the issue.

[image]

Delete that file and upload a newer version from the 8.x bundle to /lib and should be good to go.  Was unaware of this issue.. I'm sure we'll definitely see it pop up again someday.

dhalbert, 2023-08-21:
tagging @UnexpectedMaker

UnexpectedMaker, 2023-08-21:
> This was an otherwise new S2 I had in a drawer for a long time. I flashed to 8.2.3 and it no longer ran the sample code that flashes the dotstar. At this point /lib was empty, so I loaded the entire bundle into /lib. Still broken at that point. You now have the entire history.
> […](#)

Key point here is "had it for a long time" and now just re-flashed it.  The shipping code that was on there was for a much older version of the files, and new shipping files are available (as always) from here...
https://github.com/UnexpectedMaker/feathers2/tree/main/FeatherS2/CircuitPython

gmparis, 2023-08-21:
I never looked in the top level. Duh. When I went to 8.2.2, I did an erase,
which must have eliminated the unwanted .py file, correcting the issue.

Thank you for your assistance.

dhalbert, 2023-08-21:
Thanks @UnexpectedMaker.

closed 2023-08-21 by dhalbert

CrackXT, 2024-04-28:
Hello CircuitPython Team,

the adafruit_dotstar.py code that is flashed during the installation of CP 9.0.4 as well as in the 9.1.0-beta1 is incompatible.

Please replace the adafruit_dotstar.py with the new version of UnexpectedMaker.

https://github.com/UnexpectedMaker/feathers2/blob/main/FeatherS2/CircuitPython/adafruit_dotstar.py

This works again with both CP versions.

With kind regards
CrackXT

UnexpectedMaker, 2024-04-28:
> Hello CircuitPython Team,
>
> the adafruit_dotstar.py code that is flashed during the installation of CP 9.0.4 as well as in the 9.1.0-beta1 is incompatible.
>
> Please replace the adafruit_dotstar.py [...]

I don't think they should do that, sorry.

What is not working with the latest Dtostar library? Maybe post your issue rather than asking Adafruit to revert they changes?

CrackXT, 2024-04-29:
The adafruit_dotstar.py in the CP 9.0.4 package (.uf2), as well as in the 9.1.0-beta1 does not seem to be compatible with CP 9. The LED is off after the update, only if you flash your (linked) code over the one from the CP 9 package, the LED is on again.

With kind regards CrackXT

jepler, 2024-04-29:
A better place to discuss problems using the Adafruit CircuitPython dotstar library is in that github repo: https://github.com/adafruit/Adafruit_CircuitPython_DotStar -- creating a new issue there with a full description of the problem you're having would be a great idea. Please include the code you're using as well as describing the behavior you see and any messages that are shown "in the REPL".

Generally, around the time each CircuitPython release is made, each frozen library is updated to match the latest released version from its repository. Occasionally this doesn't happen due to oversight, but frozen modules were updated on April 17 for the 9.1.0-beta.1 release. If something's going wrong with this update process, or it was forgotten, then this is a good place to raise the issue.

If a change is needed for the LEDs to work right, the place to offer a pull request is in https://github.com/adafruit/Adafruit_CircuitPython_DotStar

In the specific case of Unexpected Maker devices in particular, there appears to be the extra wrinkle that some devices are shipped with .py and/or .mpy files loaded in the CIRCUITPY drive that can take precedence over the "frozen" version even after you update circuitpython, so do check if there are other dotstar "py" or "mpy" files on your device; you can also do a `storage.erase_filesystem()` to totally erase the CIRCUITPY drive (including your code.py, so back it up); after this step the "frozen" version should be the one that import finds.