adafruit/circuitpython #8307: Pico DV Base image fail to show CIRCUITPY drive or Serial for REPL.

dglaude, 2023-08-21:
### CircuitPython version

```python
Any raspberry_pi_pico_dv_base firmware, including 8.2.3: adafruit-circuitpython-pimoroni_pico_dv_base-en_US-8.2.3.uf2
But also 8.1.0 and a few other (see below).
```

### Code/REPL

```python
none
```

### Behavior

n/a

### Description

I think there is something broken with the "Pico DV" (from Pimoroni) specific CP image (from https://circuitpython.org/board/pimoroni_pico_dv_base/ ).
I don't see the CIRCUITPY drive after flashing two of my Pico (not W) with this 8.2.3 image: adafruit-circuitpython-pimoroni_pico_dv_base-en_US-8.2.3.uf2
The Pico is not broken, I tested with two different Pico, same result.
Those two Pico do display CIRCUITPY with firmware adafruit-circuitpython-raspberry_pi_pico-en_US-8.2.3.uf2 both when plugged in the Pico DV or not connected.

### Additional information

Since I don't want to use the DVI output, but I just picked that board for the I2S DAC, I am not going to worry or try to fight this bug... but there is something I don't understand and it should be easy to replicate. In fact you don't really need the Pico DV to flash that image on a pico and see the problem. I also have "Serial Port Notifier" on my PC to see the port number... and I don't see any new port with Pico DV UF2. So it is more than just CIRCUITPY drive, no REPL either. `discotool` find no device. I tested 8.1.0, 8.2.0, 8.2.1 and they all behave the same...
Also I did plug an HDIM cable too, and I don't see a signal on screen either.

dglaude, 2023-08-21:
Reading this, I believe that the problem is known: https://forums.pimoroni.com/t/adding-circuitpython-support-for-pico-dv/22012
I wonder in what version it ever worked, to see when it started to break...

RetiredWizard, 2023-08-21:
The issue on the Pimoroni forum was that the Pico W doesn't work without a bodge to CP, the Pico was working just fine last time I tested. I'll pull the latest bits and check it out.

dglaude, 2023-08-21:
> The issue on the Pimoroni forum was that the Pico W doesn't work without a bodge to CP, the Pico was working just fine last time I tested. I'll pull the latest bits and check it out.

Yes please, have a look... I am going crazy with this, I was planning a quick sound test before sleeping and fighted for more than 30 minutes.

Now checking Tod guide and pinout of the Pico DV Base... to at least have a beep. :-)

dhalbert, 2023-08-21:
`common_hal_picodvi_framebuffer_construct()` started interpreting its width and height differently, I think. I changed the arguments in board.c to `320` and `240` from `640` and `480`, and the board is now coming up. I will post a test .uf2 in a minute.

dhalbert, 2023-08-21:
[firmware.uf2.zip](https://github.com/adafruit/circuitpython/files/12403346/firmware.uf2.zip)

dglaude, 2023-08-22:
I have not tested, and not everything is super clear for me to identify main vs 8.2.x artifacts in the S3...  but shouldn't this #8309 also be applied to main so that the problem does not reappear in 9.0.

What is the opposite of backporting... forwardporting?

dhalbert, 2023-08-22:
Fixed by #8309. @dglaude -- yes, this will be merged up to main. It's actually quite a bit easier to merge from `x.x.x` to `main`, in general. There's a PR in progress to pick up a bunch of 8.2.x changes: #8317.

PR #8309 (2023-08-22): fix framebuffer size for pimoroni dvi base
- Fixes #8307 

@dglaude @RetiredWizard

I think is due to #7922, and the code for the Pimoroni board was not updated.

Tested and working for me, with DVI output.
closed 2023-08-22 by dhalbert