adafruit/circuitpython #8313: analogio not working on DOIT ESP32 Development Board

mechanater, 2023-08-22:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.3 on 8/21/2023; DOIT ESP32 Development Board
```

### Code/REPL

```python
import analogio
from board import *

pin = analogio.AnalogIn(D32)
print(pin.value)
pin.deinit()
```

### Behavior

Resulting value is always 2819, regardless of whether pin is floating or connected to a sensor, 3v3, or GND

### Description

Tested on a few analog pins, all behave the same. Using pin D32 because no analog pins appear in dir(board)

todbot, 2023-08-22:
Many MCU ADCs need a few initial reads to start working correctly.  What happens if you change your code to something like:
```py
import analogio
import board
import time

pin = analogio.AnalogIn(board.D32)

while True:
    print(pin.value)
    time.sleep(0.2)
```

mechanater, 2023-08-22:
Using the exact code provided I'm just getting repeated readings of 2819

todbot, 2023-08-22:
Can you replace your sensor with just a potentiometer?

mechanater, 2023-08-22:
Yes, I tried that also. The pin read 2819 through the full range of the potentiometer as well.

anecdata, 2023-08-22:
Can you try 
`pin = analogio.AnalogIn(microcontroller.pin.GPIO32)`?

Do _any_ of the exposed ADC1 pins give a value?

mechanater, 2023-08-22:
None of the analog pins that I tried give any value except 2819. 

I also tried `microcontroller.pin.GPIO32` and got the same result. 

I have also tried 3 different identical boards with the same result. 

The analog pins work on MicroPython just fine.

dhalbert, 2023-08-22:
ADC2 pins work for me on a HUZZAH32 breakout (similar board, without on-board USB-serial converter). ADC1 pins are stuck at 2819. Try pin `IO14`, for instance, for now, while we debug the ADC1 issue.

This pinout shows which pins use ADC1 and which use ADC2:
https://cdn-learn.adafruit.com/assets/assets/000/112/292/original/adafruit_products_Adafruit_ESP32_Huzzah_Breakout_Pinout.png?1654548847

mechanater, 2023-08-22:
I tried ADC2 but, if I recall correctly, I get an error that the board's wlan is in use.

todbot, 2023-08-22:
I did a non-exhaustive search of which pins work for ADC on a no-name brand of ESP32 board using the same ESP-WROOM-32 module that's in the DOIT ESP32 board, running CircuitPython 8.2.3 for the DOIT ESP32 board. My sensor was a 10k pot hooked up to Gnd, 3v3, and the pin in question.  The Dxx numbers in the DOIT board def are the same as IOxx numbers of the ESP32 chips)

Works fine for ADC reads:
- D4 
- D12
- D14  
- D25
- D27
- D34
- D35 

Does not work. Allows creating of `AnalogPin` but reads constant value:
- D32
- D33

Errors with "invalid pin":
- D5
- D21
- D23

todbot, 2023-08-22:
This appears to be the pinout of the DOIT board if that's helpful.

[image]

dhalbert, 2023-08-22:
I will look and see what's different about our analog-in impl vs MicroPython and/or Arduino.

mechanater, 2023-08-22:
> I did a non-exhaustive search of which pins work for ADC on a no-name brand of ESP32 board using the same ESP-WROOM-32 module that's in the DOIT ESP32 board, running CircuitPython 8.2.3 for the DOIT [...]

Oh, interesting. I think I only tried D32, D33, and a pin from ADC1.

anecdata, 2023-08-22:
> I tried ADC2 but, if I recall correctly, I get an error that the board's wlan is in use.

Should work in Circuitpython before `wifi` is imported. 'wifi.radio.enabled = False` (then `Trye`) _may_ be enough to wrap ADC2 reads interspersed with wifi use.

mechanater, 2023-08-22:
I'm using the web interface to program the ESP32, so wifi must be enabled. The entire purpose of using this board is to make use of the wifi to send sensor data via MQTT messages, so even if the analogio works without wifi enabled, it still doesn't serve the purpose I was intending. Thank you though.

mechanater, 2023-08-22:
I just tried `D34` and `D35`, and both report nothing but `2819`.

anecdata, 2023-08-22:
D34 and D35 are ADC1, so wifi shouldn't matter.

mechanater, 2023-08-22:
Also I can verify that D34 and D35 work with digitalio.

toxicantidote, 2023-11-08:
Hmm, I've been doing some testing with this. It seems that a possible trigger for this is trying to use the ADC on D32 and D33 at the same time, but I can't reproduce this consistently.

dhalbert, 2023-11-13:
Not fixed in 9.0.0-alpha.4: tested on a HUZZAH32 breakout, pin 32.

dhalbert, 2023-11-14:
This works in MicroPython with GPIO32, so I'll see what they are doing differently.

dhalbert, 2023-11-15:
I found an error in the pin table for IO32: it had the wrong ADC1 channel. The other pins are OK.

2819 is the value that corresponds to a grounded pin: a read of zero.. This is the result of looking up an actual read value of 0 in the calibration LUT (look up table).

After correcting the IO32 channel value, I was able to read reasonable values on variable ADC1 pins. It is a mystery to me why pins other than IO32 were not working for other people. I understand why IO32 was not working. I will make a PR you can test.

toxicantidote, 2023-11-15:
It was a late night with much frustration, so I'm prepared to accept my
code may also be faulty :)

Looking forward to the PR

dhalbert, 2023-11-15:
Artifacts to test are available here: https://github.com/adafruit/circuitpython/actions/runs/6871521986?pr=8607. Scroll down the page a bit.

dhalbert, 2023-11-16:
Fix now incorporated into CircuitPython 8.2.8. I'm closing this issue. Please reopen or open a new issue if you find other ESP32 `AnalogIn` problems on certain pins. 

Note that 2819 is a valid reading for a grounded pin. This is due to the calibration lookup table used to try to linearize the output.

PR #8607 (2023-11-16): Esp32 AnalogIn IO32 fix
Fixes #8313.

- Fix ADC channel for IO32.
- Use better constant names when checking pin validity.

Tested on a Feather HUZZAH32 breakout with a potentiometer and connecting to GND and 3.3V. Tested pins IO32, IO33, and IO35.
closed 2023-11-16 by dhalbert