adafruit/circuitpython #8315: Keypad Library: allow for arrays for `pull` and `value_when_pressed`

s-light, 2023-08-22:
split from https://github.com/adafruit/circuitpython/issues/7919#issuecomment-1560795024

allow for arrays for `pull` and `value_when_pressed` - 
so we can use the same object for all three buttons - 
i came across this need for the [Adafruit ESP32-S3 Reverse TFT Feather](https://www.adafruit.com/product/5691)
as on these Button D0 is opposite to D1 & D2 for the sake of deep-sleep wake-up..

this is found on the following boards:
- [Adafruit ESP32-S2 Reverse TFT Feather](https://www.adafruit.com/product/5345) [pinout - button description](https://learn.adafruit.com/esp32-s2-reverse-tft-feather/pinouts#d1-and-d2-buttons-3137135)
- [Adafruit ESP32-S3 Reverse TFT Feather](https://www.adafruit.com/product/5691)  [pinout - button description](https://learn.adafruit.com/esp32-s3-reverse-tft-feather/pinouts#d1-and-d2-buttons-3138958)

maybe there are other boards with similar setups...

Neradoc, 2023-08-22:
This might help: I wrote a library to mix multiple keypad scanner instances (or compatible) into a single event queue, with this example for using it with the reverse TFT feather:

https://github.com/Neradoc/Circuitpython_Multi_Keypad/blob/main/examples/multi_keypad_reverse_tft_feather.py

dhalbert, 2025-02-01:
Could be added via #9845