adafruit/circuitpython #8316: Keypad Library: extend for shiftregister and touchio inputs

s-light, 2023-08-22:
split from https://github.com/adafruit/circuitpython/issues/7919#issuecomment-1561834487

allow for  digitalio.DigitalInOut, adafruit_tca8418.DigitalInOut,  adafruit_74hc595.DigitalInOut and the like.
This way we can remove shift register scanner from keypad library altogether without losing shift register support and still gain support for shift register or mixed matrices.

usage idea - all these examples should work..

based on [tutorial one button](https://learn.adafruit.com/key-pad-matrix-scanning-in-circuitpython/keys-one-key-per-pin#one-button-example-3099040)  [code on github](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Keypad_Examples/one_button/code.py)
(current behavior - the keys library sets up the pins)
```python
import board
import keypad

KEY_PINS = (
    board.D5,
    board.D6,
)

keys = keypad.Keys(
    KEY_PINS, 
    value_when_pressed=False, 
    pull=True
)
event = keypad.Event()

while True:
    keys.events.get_into(event):
    if event:
        print(event)
```

loosely based on https://learn.adafruit.com/74hc595/usage
(new concept shiftregister)
```python
import time
import board
import digitalio
import adafruit_74hc595

latch_pin = digitalio.DigitalInOut(board.D4)
sr = adafruit_74hc595.ShiftRegister74HC595(board.SPI(), latch_pin)
sr_pins = [sr.get_pin(n) for n in range(8)]

KEY_INPUTS = (
    *sr_pins,
)

keys = keypad.Keys(
    KEY_INPUTS, 
    value_when_pressed=False,
)
```

extended by https://learn.adafruit.com/adafruit-qt-py-esp32-s3/capacitive-touch#reading-touch-on-the-pins-3095937
```python
import time
import board
import touchio
import digitalio
import adafruit_74hc595

# simple pins
button_pin1 = digitalio.DigitalInOut(board.D5)
button_pin1.switch_to_input(pull=digitalio.Pull.UP)
button_pin2 = digitalio.DigitalInOut(board.D6)
button_pin2.switch_to_input(pull=digitalio.Pull.DOWN)

# shift register
latch_pin = digitalio.DigitalInOut(board.D4)
sr = adafruit_74hc595.ShiftRegister74HC595(board.SPI(), latch_pin)
sr_pins = [sr.get_pin(n) for n in range(8)]

# touch
touch_one = touchio.TouchIn(board.A2)
touch_two = touchio.TouchIn(board.TX)

KEY_INPUTS = (
    *sr_pins,
    button_pin1,
    button_pin2,
    touch_one,
    touch_two,
)

keys = keypad.Keys(
    KEY_INPUTS, 
    value_when_pressed=(
        8*(False,),
        4*(True,),
    ),
)

keys = keypad.Keys(
    KEY_INPUTS, 
    value_when_pressed=False,
)
```

open point: to safe space it would be ideal to remove the special shiftregister version from the lib...
but i do not know if this is *possible* at all from a *usage & documentation* point of view...

jepler, 2023-08-22:
I think there are some good ideas here! Thank you for raising this feature request.

That said, I'd like to give some context on why this is implemented as it is.

It's worth acknowledging that the implementation that is ONLY compatible with a `microcontroller.Pin` object is certainly the easiest implementation! and that is no doubt part of why it's that way.

Another less obvious thing about `keypad` is that the scanning happens in what we call the "background task" context. At the moment this is not truly a "task" in the sense of an RTOS, but it does allow code to be executed while CircuitPython is otherwise unable to run foreground Python code, usually because it's performing a blocking operation of some sort.

While there are already ways to call out to arbitrary Python code from the background task (e.g., when `adafruit_sdcard` is used, rather than the newer and preferred `sdcardio`), we have found that this is problematic and don't want to introduce new cases where this happens (e.g., see #2417) unless we can understand why it is "safe" to do so.

I also want to comment in particular about potentially using `adafruit_74hc595.DigitalInOut` to replace `ShiftRegisterKeys`. I think this is sub-optimal: Based on a quick reading the source of `adafruit_74hc595` it appears that each use of `DigitalInOut.value` reads the whole shift register, so if there are 16 keys it would read the shift register 16 times for a total of 256 bits shifted...

Out of all the ideas presented here, I think the candidate for "most implementable" out of these ideas would be to permit core `TouchIn` objects to be passed in to a Keys, or to a new TouchKeys class. This would be quite handy & useful for a lot of people.

When it comes to combining different sorts of inputs, another approach I have considered would be to add some new arguments to the constructors, `queue=None, key_number_offset=0`. Multiple input sources could drop events in the same queue object, with the offset applied so that the key numbers remain distinct.

dhalbert, 2023-08-22:
@jepler Thanks for the comments.

A few more comments:
`keypad` is implemented in C to do the background scanning and debouncing. The debouncing in particular requires low latency. For `TouchIn`, and maybe even shift registers, scanning could instead be done in CircuitPython code. Right now there isn't a clean way to do this, but if the `EventQueue` were made more general so that events could be added from Python, I think that would help. Then we could add `keypad`-style libraries coded in CircuitPython. Perhaps the shift register code could even be moved out of C. (There might or might not be latency issues for people who want fast game controllers.)

deshipu, 2023-08-22:
I think that at least for touch, the C code could be implemented much better than the naive python code would do it. I still have https://github.com/adafruit/circuitpython/issues/6445 on my todo list somewhere, but I doubt I will be able to look at it this year.

The idea is simple: measuring capacitance for touch requires a certain delay – that makes checking all the pads one by one with python code individually rather slow. But many platforms that have support for touch allow you to initiate the test for multiple pins at once, wait only once, and check the results in bulk. Some even let you to do a touch matrix, which would greatly increase the number of pads you can have.

s-light, 2023-08-22:
thanks for your comments!

ok - so to summarize:
the idea in itself is good - 
only not implementable at the current CP state as there are to much edge-cases that can create big trouble...

as i originally came from the need for touchio - i will follow https://github.com/adafruit/circuitpython/issues/6445 more closely...