adafruit/circuitpython #8321: Add a git tag and/or PR number to artifact and PR merge build filenames

dhalbert, 2023-08-22:
When we merge a PR, firmware are uploaded to S3. Before that, artifacts are available.
The files have names like `adafruit-circuitpython-8086_commander-en_GB-20230803-95c0867.uf2`. This is not all that helpful. At the moment, when we are making both `main` and `8.2.x` PR's, it would be nice if the names included the nearest tag and possibly the PR number.

anecdata, 2023-08-22:
Seconded. PR would be particularly efficient for bisects.

tannewt, 2023-08-24:
Version name is determined here: https://github.com/adafruit/circuitpython/blob/main/tools/build_board_info.py#L61

dhalbert, 2023-11-24:
This worked for the PR builds, but the merge build has an incorrect name, e.g.
`adafruit-circuitpython-8086_commander-de_DE-20231124-no-branch-3aac226.uf2`. So needs more work.

jepler, 2023-11-24:
The following git command will show the current branch name (or show "HEAD" if you currently have a 'detatched head'): `git rev-parse --abbrev-ref HEAD`

dhalbert, 2023-11-24:
> The following git command will show the current branch name (or show "HEAD" if you currently have a 'detatched head'): `git rev-parse --abbrev-ref HEAD`

I was doing something similar, but `GITHUB_BASE_REF` was the same thing. I expected that value to be available on the merge build, but apparently not. In addition, the `PULL` value appears to not be present, and I thought it was the same build step. I'll need to re-check those.

dhalbert, 2024-02-16:
Fixed by #8930.

PR #8648 (2023-11-24): add branch and PR number to uploaded build filenames
Fixes #8321.

Adds more information to uploaded build filenames, e.g.:
was
```
adafruit-circuitpython-adafruit_feather_esp32s2-en_US-20231122-b2c32cf.uf2
```
will be
```
adafruit-circuitpython-adafruit_feather_esp32s2-en_US-20231122-main-PR8645-b2c32cf.uf2
```
closed 2024-02-16 by dhalbert