adafruit/circuitpython #8324: Current main tagging is stuck at 8.2.0

bill88t, 2023-08-23:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.0-87-ge08ad22ef on 2023-08-22; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
N/A
```

### Behavior

It should be 8.2.4-...

dhalbert, 2023-08-23:
`8.2.x` was split off from `main` at 8.2.0, so the tag would not advance.

I am going to tag with a 9.0.0 tag soon. This does not mean we are going to make a 9.0.0 release. We will wait until the .mpy version stabilizes. It is changing at least twice as we merge from MicroPython. The v1.19.1 merge already made one change.

closed 2023-08-23 by dhalbert

dhalbert, 2023-08-23:
If you do `git pull --tags` from `adafruit/circuitpython` now, you should get a
`9.0.0-alpha.0` tag (first PR after 8.2.0 split)
and a
`9.0.0-alpha.1` tag (MicroPython v1.19.1)