adafruit/circuitpython #8328: Keypad and pn532 don't play nice together

crazyiop, 2023-08-23:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.4 on 2023-08-22; BPI-PicoW-S3 with ESP32S3
```

### Code/REPL

```python
import board
import busio
import sdcardio
import storage
import audiobusio
import digitalio
import keypad
import gc
from adafruit_pn532.spi import PN532_SPI
import time

#########
# Buttons
#########

keys = keypad.Keys(
    [board.GP13, board.GP14, board.GP15, board.GP16, board.GP17],
    value_when_pressed=False,
    pull=True
)

#########
# sd card
#########

do = board.GP9
sck = board.GP10
di = board.GP11
sd_cs = board.GP12

spi = busio.SPI(sck, MOSI=di, MISO=do)
sdcard = sdcardio.SDCard(spi, sd_cs)

vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
print('sd card ready')

#####
# nfc
#####

sck = board.GP18
miso = board.GP19
mosi = board.GP20
nfc_cs = board.GP21

spi = busio.SPI(sck, mosi, miso)
cs_pin = digitalio.DigitalInOut(nfc_cs)
pn532 = PN532_SPI(spi, cs_pin, debug=False)
pn532.SAM_configuration()
print('PN532 ready !')

#####
# I2S
#####

# using a max98357 breakout
# https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/overview
# i2s amp power direcly connected to the 5V of the board (Usb power)

din = board.GP3
bclk = board.GP4
lrclk = board.GP5
audio = audiobusio.I2SOut(bit_clock=bclk, word_select=lrclk, data=din)
print('i2s ready')

##############
# Player logic
##############

while True:
    playing_uid = None
    print('waiting for a tag to play·...')
    while playing_uid is None:
        playing_uid = pn532.read_passive_target(timeout=0.1)
        if playing_uid:
            break
        time.sleep(1)
        gc.collect()

        # this loop eventually crash when doing nothing (not puting a tag in
        # range, nor pressing any of the button of the Keypad)

    print(playing_uid)
    print('part where I play a song from sd depending on the nfc tag ID...')
```

### Behavior

I encouter crash with the message `You are in safe mode because: Internal watchdog timer expired.` and get that everytime I run the code.

It started to happen When I added a 5 key `Keypad` to my previously working project (a player that play song from an sdcard upon finding an nfc tag). 

The project, at a previous state, was working fine with: nfc poling + i2s song playing + some control from an i2c rotary encoder poling (volume, next on click). I wanted to move from the rotary to a `Keypad` for the controls.

### Description

The given code is a minimal failing example of my code.
- as is, it will crash, from ~10s to ~2min (but rarely more than 1min)
- When commenting the `pn532.read_passive_target()`: It works fine (no crash after 10+ minutes)
- When commenting the whole `Keypad` definition: It works fine too (no crash after 10+ minutes)

tannewt, 2023-08-24:
Probably a duplicate of #8288

crazyiop, 2023-08-24:
@tannewt I just try and I can run the more minimal code of this issue without error on my board (I'm in 8.2.4, Her test was at in 8.2.0-75)

tannewt, 2023-08-28:
I don't think this is a duplicate of #8288 because this board uses 2MB PSRAM which is in quad mode (not octal.)

tannewt, 2023-08-29:
Please try this with artifacts from #8345 because it'll detect pin issues that can lead to this error. This example may have a similar problem.

crazyiop, 2023-08-29:
It doesn't seems to help for me...
```
Adafruit CircuitPython 8.2.4-4-g811bd12fa on 2023-08-29; BPI-PicoW-S3 with ESP32S3
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
sd card ready
PN532 ready !
i2s ready
waiting for a tag to play·...

<serial broke>

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

crazyiop, 2023-08-29:
after some minor change (simple refactoring, trying to make my failing code example even more minimal), I managed to get another error:
```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

tannewt, 2023-08-29:
The hard fault will be easier to debug! Please post the minimized code. Thanks!

crazyiop, 2023-08-29:
Actually, the code differences don't matter, I now get both error apparently randomly (but more frequently the hard fault) also with the code previously given...

tannewt, 2023-08-30:
Do you have any Adafruit or Espressif dev boards you can reproduce this on? I don't have a BPI. Alternatively, you could try building a debug build and getting a backtrace from the debug uart output.

crazyiop, 2023-08-30:
I do have :
- an esp32-c3 qt py 
- a magtag (esp32-s2)
- a circuit playground express (not esp but that would be my test group)

but all of them would require a bit of soldering/desoldering with my current code.

I'm currently trying to get a crashing code that does not need actual hardware connected. I mostly still need to get rid of the nfc pn532. I tried to replace the `pn532.read_passive_target()` with spamming somewhat similar spi command but it does not crash. So my next step is to copy the whole spi PN532 class and tweak it to behave similarly in my case with assuming the actual hardware reply...

I'm almost there (I hope) but I'm working slowly on it as I do not have that much spare time... But I do want to get to the bottom of this !

I had not considered using a debug build yet. I feel that it would take me longer than the above tests so I would try that afterwards.