adafruit/circuitpython #8337: MAIN branch build fails for port RP2

eightycc, 2023-08-27:
### CircuitPython version

```python
Current top of MAIN
```

### Code/REPL

```python
make V="steps commands rules" BOARD=raspberry_pi_pico
```

### Behavior

supervisor/internal_flash.c:35:10: fatal error: genhdr/flash_info.h: No such file or directory

### Additional information

Seems to be a problem in the port Makefile where it fails to run gen_stage2.py.

eightycc, 2023-08-27:
Comparing MAIN with tag 8.2.4 there are significant changes to way that the qstr collection is built. On MAIN, py/makeqstrdefs.py runs first to generate genhdr/qstr.i.last, but it fails because supervisor/internal_flash.c includes genhdr/flash_info.h which has not been generated yet. So, this rule (or similar) in ports/raspberrypi/Makefile:

```
$(HEADER_BUILD)/flash_info.h: flash_info.h.jinja gen_stage2.py | $(HEADER_BUILD)/
	$(STEPECHO) "GEN $<"
	$(Q)$(PYTHON) gen_stage2.py $< $@ $(EXTERNAL_FLASH_DEVICES)
```

needs to run before makestrdefs.py.

eightycc, 2023-08-27:
Worked around the problem by making the include of genhdr/flash_info.h dependent on NO_QSTR like this:

```
#ifndef NO_QSTR
// Only include flash_info when not doing qstr extraction.
#include "genhdr/flash_info.h"
#endif
```

Not sure if this is most desirable fix, but let me know if it is and I'll make a pull.

dhalbert, 2023-08-27:
@jepler: your fix seems useful, but why was this not failing in the CI or locally before?

jepler, 2023-08-27:
I don't know why it wasn't seen during CI but it reproduced locally with
```
make BOARD=raspberry_pi_pico clean && make BOARD=raspberry_pi_pico build-raspberry_pi_pico/genhdr/qstr.i.last
```

eightycc, 2023-08-28:
Successfully tested pull #8340 locally.

PR #8340 (2023-08-28): Use QSTR_GLOBAL_REQUIREMENTS to give explicit dependency for qstr gen…
…eration

I investigated and found that this seems to be the best way to specify a file that must be generated before "qstr processing" can occur.

Closes #8337
commit b0967cb1 (2023-08-28): Use QSTR_GLOBAL_REQUIREMENTS to give explicit dependency for qstr generation

I investigated and found that this seems to be the best way to specify
a file that must be generated before "qstr processing" can occur.

Closes #8337