adafruit/circuitpython #8353: Protomatter library updates

PaintYourDragon, 2023-08-31:
Protomatter matrix library 1.6.1 (c9c1189) addresses some issues with ESP32-S3. Drive strength compilation issue is fixed, pixel clock is slightly reduced to help w/visual glitches.

The prior release, 1.6.0, if not already in use by CircuitPython, addresses M4 issues. Also a slight pixel clock reduction, and additionally the clock duty cycle is semi-configurable -- perhaps this could be exposed as a named arg in Python, and ignored if not supported by arch. See _PM_setDuty() comments in that lib. Thx.

tannewt, 2023-09-05:
@dhalbert want to do this before 8.2.5?

dhalbert, 2023-09-06:
> @dhalbert want to do this before 8.2.5?

I think so, but maybe not expose the argument (yet).

dhalbert, 2023-11-07:
Fixed by #8376.

PR #8376 (2023-09-07): update protomatter to 1.6.1 (used for RGBMatrix)
- Fixes #8353, updating protomatter to the latest set of fixes
closed 2023-11-07 by dhalbert