adafruit/circuitpython #8359: Error with USB Keyboard Input Read

berto-1, 2023-09-02:
Hi everyone,

I would like to connect a keyboard to a Raspberry Pico (RP2040) with the data+ and data- cables of the USB and check the content of an input string. 
From the circuitpython forum I know that recently it has been implemented the usb_host library to use the PIO of the Pico and I tried using my Pico W board with the following script:

```
import board
import usb_host
import sys

h = usb_host.Port(board.GP0, board.GP1)

while True:
    print(sys.stdin.read(1))
```

However I obtained the following error, which could be due to a conflict with PIO used by the communication to the wifi chip

Traceback (most recent call last):
File "<stdin>", line 5, in <module>
RuntimeError: All state machines in use

juergenpabel, 2023-09-02:
Hi,

I just started with CP9.0.0 and usb_host on my PicoW yesterday and also came across that issue. I fixed it by switching the Pico PIO state-machines for USB D+ and D- and got it running....sort of. Here's the diff:
```
diff --git a/ports/raspberrypi/common-hal/usb_host/Port.c b/ports/raspberrypi/common-hal/usb_host/Port.c
index 93d19acd6..6b9aeae55 100644
--- a/ports/raspberrypi/common-hal/usb_host/Port.c
+++ b/ports/raspberrypi/common-hal/usb_host/Port.c
@@ -122,8 +122,8 @@ usb_host_port_obj_t *common_hal_usb_host_port_construct(const mcu_pin_obj_t *dp,
     pio_usb_configuration_t pio_cfg = PIO_USB_DEFAULT_CONFIG;
     pio_cfg.skip_alarm_pool = true;
     pio_cfg.pin_dp = dp->number;
-    pio_cfg.pio_tx_num = 0;
-    pio_cfg.pio_rx_num = 1;
+    pio_cfg.pio_rx_num = 0;
+    pio_cfg.pio_tx_num = 1;
     // PIO with room for 22 instructions
     // PIO with room for 31 instructions and two free SM.
     if (!_has_program_room(pio_cfg.pio_tx_num, 22) || _sm_free_count(pio_cfg.pio_tx_num) < 1 ||
```
It seems that the other PIO(and thus: state-machine)-using components filled up that one PIO so much it wouldn't fit anymore - but apparently the other PIO was still plenty available and just switching them for their rx/tx duties seems to work (at least no more warnings/errors/exceptions).

**But**: with that patch it doesn't complain anymore ....but it doesn't find any devices when running
`usb.core.find(find_all=True)`
(in a loop with sleeps in between). I tried various USB devices and none were "found" by the Pico (using GP27 for D+ and GP26 for D-); power is provided to my attached (other) USB devices but they also seem to also not see their USB host (the Pico on GP26+GP27). 

It "feels" as if the USB PIO logic wasn't (correctly?) started by CP - a quick glance at that part of the code did look OK though.

Maybe @tannewt, @dhalbert or @jepler have some ideas? I'll probably get back to this on monday.

And just for reference the relevant excerpt from my code:
```
usb_host.Port(board.GP26, board.GP27)
if supervisor.runtime.usb_connected:
  print("USB<host>!")
else:
  print("!USB<host>")
while True:
    print(usb.core.find())
    time.sleep(1)
```

juergenpabel, 2023-09-04:
I had a combination of different causes that at led to my issue:
- a not so clean solder-point on the D+ cabling => more solder for a more reliable connection
- a very nitpicky USB device (a Micro RP2040 from SBcomponents) => other boards work just fine (that SBcomponents board works just fine with the examples from Pico-PIO-USB - so it's something within circuitpython that behaves different)
- invoking usb_host.Port(...) in boot.py doesn't work => invoking it from code.py does work

Not relevant (but tested) were:
- setting cpu#1 frequency to 120mhz (instead of CP default 125mhz) to be a multiple of 12mhz (USB low speed)
- disabling picodvi and other PIO-consumers
- switching to other GPIO pin-pairs for D+/D-
- time.sleep(1) at various locations in the code
 
The only (just mildly bothersome) issue I have now is that the first enumeration always causes this error:
```
[1:0] Control FAILED, xferred_bytes = 8
80 06 00 01 00 00 08 00 
Enumeration attempt 1
```
...but since my logic is inside a `while True` anyhow, all subsequent enumerations just work fine. So I'm happy (for now).

Conclusion: switching the PIO engines for RX and TX (diff from above) solved my issue with CP9.0.0-HEAD and a PicoW. Should I do a PR or will someone pick it up?

tannewt, 2023-09-05:
I'm fine to see the default switched. It should be fine either way.

juergenpabel, 2023-09-05:
> Conclusion: switching the PIO engines for RX and TX (diff from above) solved my issue with CP9.0.0-HEAD and a PicoW.

Just for reference (in case anybody is interested): https://github.com/juergenpabel/circuitpython-picow-escpos/

berto-1, 2023-09-16:
I uploaded the last uf2 file to my PicoW and the previous error related to the pio machines seems to be solved by using the GP26 and GP27.

My code is:

```
import microcontroller
microcontroller.cpu.frequency = 120000000

print(microcontroller.cpu.frequency)

import board
import usb_host
import usb
import sys
import supervisor
import time

usb_host.Port(board.GP26, board.GP27)
if supervisor.runtime.usb_connected:
  print("USB<host>!")
else:
  print("!USB<host>")
  
while True:
    print(usb.core.find())
    print("Keyboard Reading...")
    print(sys.stdin.read(1))
    time.sleep(1)
```

and my output

120000000
USB<host>!
<Device>
Keyboard Reading...

But it seems that it is not able to read and print any press of buttons on the keyboard. I am new to circuitpython and maybe there is a different way of reading the usb.
Thanks.

juergenpabel, 2023-09-16:
> But it seems that it is not able to read and print any press of buttons on the keyboard. I am new to circuitpython and maybe there is a different way of reading the usb.

I would suspect that the USB keyboard isn't assigned/added to the stdin-"channel" automatically - all search results I got refer to the board emulating a USB keyboard (and not using a keyboard for stdin as a USB host). But some code parts in supervisor/shared/usb/host_keyboard.c seem to implement just that....maybe I find some time next week to look into this further.

tannewt, 2023-09-18:
Does it work the second time you run the code? I'm not sure the workflow code runs while user code is running.

What keyboard model is it?

abv, 2024-01-17:
Circuitpython 9.0.0-alpha.6 gave me the `RuntimeError: All state machines in use` error on my Raspberry Pico W. After building circuitpython based on the diff provided [above](https://github.com/adafruit/circuitpython/issues/8359#issuecomment-1703962079), I was able to get things working.

jepler, 2024-01-31:
@abv @berto-1 could one of you prepare that change as a pull request?

commit 3bb5c26a (2024-02-15): Swap PIO allocation of USB Host for better Pico W compatibility

Closes: #8359
closed 2024-02-15 by tannewt