adafruit/circuitpython #8366: Using busio.UART on nRF breaks USB MSD

caternuson, 2023-09-05:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.4 on 2023-08-22; Adafruit ItsyBitsy nRF52840 Express with nRF52840
```

### Code/REPL

```python
This code demonstrates the issue:

import busio
import board
import time

uart_hw = busio.UART(board.TX, board.RX)

while True:
    print("Spin Loop")
    time.sleep(0.5)

For comparison, this code works OK. The only difference is the use of the `board.UART()` singleton:
```python
import board
import time

uart_hw = board.UART()

while True:
    print("Spin Loop")
    time.sleep(0.5)
```
```

### Behavior

The looping and printing works as expected for both code examples. The issue lies more with the subsequent effect on USB behavior. More info in Description below.

### Description

With the first code example above loaded as `code.py`, it runs as expected with a soft reset:
```python
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Spin Loop
Spin Loop
Spin Loop
Spin Loop
Spin Loop
Spin Loop
Traceback (most recent call last):
  File "code.py", line 9, in <module>
KeyboardInterrupt: 

Code done running.
```
However, after breaking out via <CTRL><C>, the REPL does not come back. At this point, the board is now in a weird/bad state. Attempting to alter files in `CIRCUITPY` fails. So can not recover by simply changing or deleting the offending `code.py`. Must get into safemode and perform a `storage.erase_filesystem()`, or something equivalent.

For the code that uses the `board.UART()` singleton, the behavior is as expected:
```python
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Spin Loop
Spin Loop
Spin Loop
Traceback (most recent call last):
  File "code.py", line 8, in <module>
KeyboardInterrupt:

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.2.4 on 2023-08-22; Adafruit ItsyBitsy nRF52840 Express with nRF52840
>>> print("hello")
hello
>>>
```
The code loops and prints. Hitting <CTRL><C> breaks out and drops into REPL. The `CIRCUITPY` drive still behaves OK, etc. No issues.

### Additional information

Related forum thread:
https://forums.adafruit.com/viewtopic.php?p=983473

dhalbert, 2023-09-12:
This started being an issue between 8.2.0-beta.1 and 8.2.0-rc.0. Investigating futher.

dhalbert, 2023-09-12:
@furbrain Sorry to report that this issue is being caused by #8089, your UART power-saving PR. If I revert that the problem goes away. It appears that power-cycling the UART peripheral leaves something in a bad state.

For the next 8.2.x release, I am going to just revert #8089 for now, even though that will cause increased sleep current. It would be great to figure out the best way to do this.

dhalbert, 2023-09-12:
Fixed by #8391.

PR #8391 (2023-09-12): 8.2.x revert #8089 nrf uart power saving
- Fixes #8366.

It would be nice to rework #8089 in the long run, since it does address a sleep power problem.

I had to manually revert the commits to do this on the 8.2.x branch. Pressing the "revert" button and changing the base did not work out.

Tagging @furbrain for interest.
closed 2023-09-12 by dhalbert