adafruit/circuitpython #8368: LYWSD02 Support Request (Crashed hard trying to use it as is)

Axeia, 2023-09-05:
I figured I'd give this a shot as at least from the outside the LYWSD02 seems awfully similar. It displays the same data and it seems to share the same UUID https://github.com/adafruit/Adafruit_CircuitPython_BLE_LYWSD03MMC/blob/8bc79c6fcb72699c159129e8a1b0d27a5eaec9f7/adafruit_ble_lywsd03mmc.py#L66

So I put `adafruit_ble_lywsd03mmc.py` in my libs directory and copied over the content from `ble_lywsd03mmc_simpletest.py` to my main.py.
The only thing I changed was change line 19
https://github.com/adafruit/Adafruit_CircuitPython_BLE_LYWSD03MMC/blob/8bc79c6fcb72699c159129e8a1b0d27a5eaec9f7/examples/ble_lywsd03mmc_simpletest.py#L19

to 
```python
        if adv.complete_name == "LYWSD02":
```

Ran the code and I got this in the terminal:
```
Press any key to enter the REPL. Use CTRL-D to reload.
  File "main.py", line 28, in <module>
  File "/lib/adafruit_ble/__init__.py", line 104, in __getitem__
  File "/lib/adafruit_ble/__init__.py", line 68, in _discover_remote
MemoryError: Nimble out of memory

Code done running.
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
[Open] Connection to COM6

press Ctrl-C to enter the REPL

soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Crikey!
Heap allocation when VM not running.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.
```

I'm guessing this is not actually due to a shortage of RAM although I don't have a board with more RAM to test it on.
I'm using a [BPI-Leaf-S3](https://wiki.banana-pi.org/BPI-Leaf-S3) which is an ESP32-S3 based board with this memory configuration:
| Type              | Amount |
|-------------------|--------|
| Internal SRAM     | 512 KB |
| Onboard FLASH ROM | 8MB    |
| In-packge PSRAM   | 2MB    |

If it's of any help to add support there seems to be a PyPi library for the LYWSD02 over here:
https://github.com/h4/lywsd02

[Adafruit actually seems to stock](https://www.adafruit.com/product/5023) this particular Clock/thermometer/hygrometer so support for it would be neat.

I'm very new to circuitpython but let me know if you need more info :)

tannewt, 2023-10-23:
What version of CircuitPython was this with? Please try "Absolute Newest" to see if it is fixed there. Thanks!

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.