adafruit/circuitpython #8370: espressif web workflow hard fault on reload after wifi.radio.enabled = False

anecdata, 2023-09-06:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.4 on 2023-08-22; Adafruit QT Py ESP32-S3 no psram with ESP32S3
```

### Code/REPL

```python
import os
import supervisor
import wifi

print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")
if os.getenv("WIFI_SSID") and not wifi.radio.connected:
    print(f"Connecting...")
    wifi.radio.connect(os.getenv("WIFI_SSID"), os.getenv("WIFI_PASSWORD"))
print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")

# espressif: hard fault on 2nd run if web workflow is enabled
# raspberrypi: can't test, wifi.radio.enabled not fully implemented
# wifi.radio.enabled = False

# espressif: OK* on successive runs regardless of web workflow
# raspberrypi: OK on successive runs
wifi.radio.stop_station()

# *when using wifi.radio.stop_station()
#   wifi.radio.ipv4_address=None will lag slightly to wifi.radio.connected=False (code.py, but not REPL)
#   Connected=False IP=0.0.0.0
#   or
#   Connected=False IP=192.168.6.210
print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")
while wifi.radio.ipv4_address:
    print(".", end="")
print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")
time.sleep(1)
supervisor.reload()
```

### Behavior

After 1st reload, when using `wifi.radio.enabled = False`:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

bill88t, 2023-09-06:
Umm. This is a scenation I not only tested, but also works on my boards..

Simplifying to:
```py
import wifi
import supervisor
from time import sleep

wifi.radio.enabled = 0

sleep(3)
supervisor.reload()
```
Does not produce any hardfaults..

And the example code needs fixing:
```py
import os
import supervisor
import wifi
from time import sleep

print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")
if os.getenv("CIRCUITPY_WIFI_SSID") and not wifi.radio.connected:
    print(f"Connecting...")
    wifi.radio.connect(os.getenv("CIRCUITPY_WIFI_SSID"), os.getenv("CIRCUITPY_WIFI_PASSWORD"))
print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")

# espressif: hard fault on 2nd run if web workflow is enabled
# raspberrypi: can't test, wifi.radio.enabled not fully implemented
wifi.radio.enabled = False
#
# or
#
# espressif: OK* on successive runs regardless of web workflow
# raspberrypi: OK on successive runs
# wifi.radio.stop_station()

# *when using wifi.radio.stop_station()
#   wifi.radio.ipv4_address=None will lag slightly to wifi.radio.connected=False (code.py, but not REPL)
#   Connected=False IP=0.0.0.0
#   or
#   Connected=False IP=192.168.6.210
print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")
while wifi.radio.ipv4_address:
    print(".", end="")
print(f"Connected={wifi.radio.connected} IP={wifi.radio.ipv4_address}")
sleep(5)
supervisor.reload()
```

And still doesn't hardfault.

I unfortunately can't reproduce on S3.

anecdata, 2023-09-06:
Probably not necessary, it's a bit if an edge case.

dhalbert, 2023-09-12:
> Shall I create a minimal "backport" just doing the fix?

@bill88t do you mean this is fixed in `main`?
@tannewt did you want this fixed in 8.2.x?

tannewt, 2023-09-12:
I don't really care. I think I put it in 8.2.x because it is a crash. I don't want to fix it myself atm though.

bill88t, 2023-09-12:
@dhalbert  This was a crash I found as I was seperating wifi-autoconnect from web-workflow. In the rush of things, as I was testing edge cases, I assumed it's a side effect of my changes and fixed it. It went in main with that pr.
After seeing this issue however, I realized it was broken all along.
I offered to backport just the fix, but since it was deemed unimportant, I took no further action.

dhalbert, 2023-09-12:
That's fine; we can just leave it for now.

anecdata, 2023-12-08:
Retested with `Adafruit CircuitPython 9.0.0-alpha.5 on 2023-11-15; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3`. Well-behaved on `espresso` and `raspberry`. Closing.

closed 2023-12-08 by anecdata