adafruit/circuitpython #8380: "Extra arguments" message returned if not using named arguments with sslContext.load_verify_locations

tyeth, 2023-09-08:
### CircuitPython version

```python
8.2.4 + 9.0.0-dirty from end of august (before final certs submodule change as part of resolving #8299 )
```

### Code/REPL

```python
import ssl
context = ssl.create_default_context()
data = MY_CA_DATA # some pem string or bytes from a file of single cert or bundle (was testing bundle)
context.load_verify_locations(data)
```

### Behavior

Reports extra arguments given: 1

### Additional information

Supplying the argument as a named argument allows the code to proceed 
e,g, ca_data= (or was it cadata="")

dhalbert, 2023-09-08:
It is currently required to be a keyword. But that is not true in CPython. Easy to fix.
https://github.com/adafruit/circuitpython/blob/0928a95bb203a4b879fc4b48fc2946a34a2a0458/shared-bindings/ssl/SSLContext.c#L103

dhalbert, 2023-09-12:
Fixed by #8392.

PR #8392 (2023-09-12): 82.x - Remove keyword-only requirement for SSLContext.load_verify_locations() arg
- Fixes #8380.
closed 2023-09-12 by dhalbert