adafruit/circuitpython #8381: sslContext.load_verify_locations doesn't work as expected

tyeth, 2023-09-08:
### CircuitPython version

```python
All
```

### Code/REPL

```python
import ssl
context = ssl.create_default_context()
data = MY_CA_DATA # some pem string or bytes from a file of single cert or bundle (was testing bundle)
context.load_verify_locations(data)
```

### Behavior

If a path is passed, via ca_path named argument, or a certificate (or bundle of certificates) is passed in PEM format using the named argument cadata/ca_data then it should be accepted, and the bundle should be used for new ssl connections using the created context. The bundle is accepted without raising an exception, until a request is made, then the ESP32 SDK raises an error about processing the CRT (8015 in esp tls)

```
Fetching json from https://www.adafruit.com/api/quotes.php
Traceback (most recent call last):
  File "adafruit_requests.py", line 534, in _get_socket
OSError: Unhandled ESP TLS error 10368 0 8015 -1
```

The certificate data argument does not accept a bundle. It should.
The function does not accept a file path as string, it should.

This would maintain CPython compatibility (if #8380 was fixed too)

anecdata, 2023-09-08:
(hope you don't mind, I syntax-highlighted the example code)

tyeth, 2023-09-08:
> (hope you don't mind, I syntax-highlighted the example code)

I'm truly outraged! Mainly that I didn't do it 😭

dhalbert, 2023-11-16:
I did not realize that there are three args in CPython.
```
SSLContext.load_verify_locations(cafile=None, capath=None, cadata=None)
```
So #8392 makes it _more_ incompatible with CPython.

Edited the OP to add four tasks.